simat/BatteryMonitor #2: Assistance1

harmongt01, 2017-03-02:
Hi,

The project is very close to what I'm trying to do with the same hardware.  Imported your code onto the BeagleBone Black, which went well thanks to your clear instructions and nice work.  I've reviewed your descriptions and schematic but I need a little help to modify for my project.  

Here's an example of the monitor points:
12V battery bank, voltage of entire bank.  Reading 1
12V current shunts on incoming power from solar panels.  Reading 2
12V current shunts on incoming power from wind generator1.  Reading 3
12V current shunts on incoming power from wind generator2.  Reading 4
12V current shunts on outgoing power from battery bank to inverter.  Reading 5

It looks like from the schematic you did not use a power divider for the Bat Current, is that correct?

I am new to this so I'd like to use work you've already done.  Tweak it for the above readings and go from there.

Thanks in advance,
George

harmongt01, 2017-03-03:
Tried editing the battery.cfg to what I thought looked ok for on current monitor point but am getting this error:
root@beaglebone:~/Adafruit_Python_ADS1x15/BatteryMonitor# python
Python 2.7.9 (default, Aug 13 2016, 17:56:53) 
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import batteries
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "batteries.py", line 32, in <module>
    config['alarms'][i][1] = compile(config['alarms'][i][1], '<string>', 'exec') 
TypeError: expected a readable buffer object

How can I enable a debug log?

harmongt01, 2017-03-04:
Made some progress:
>>> import batteries
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "batteries.py", line 27, in <module>
    from getdata import Readings
  File "getdata.py", line 32, in <module>
    class Readings:
  File "getdata.py", line 40, in Readings
    avoffset = avoffset/numcells
ZeroDivisionError: float division by zero
>>> batteries.deamon()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'batteries' is not defined

simat, 2017-03-04:
Can you post your battery.cfg file

harmongt01, 2017-03-04:
[battery.txt](https://github.com/simat/BatteryMonitor/files/818297/battery.txt)
 
I changed the ext to .txt so the site would accept it.

simat, 2017-03-04:
numcells should be 1.

harmongt01, 2017-03-04:
ok.  I was trying to one shunt meter using posinon AIN0 and AIN1 of only one ADS1115 to start with.

harmongt01, 2017-03-04:
My message was not clear, sorry.  What I meant to say is I was trying to start with one shunt and one ADS1115 to take a current measurement using differential inputs.

Then when that works I will add the others as well as a voltage measurement of the battery bank.

simat, 2017-03-04:
When I wrote the code I never thought about not monitoring battery voltage. I would just set up one dummy voltage input.

The code could be altered ...

harmongt01, 2017-03-04:
Thanks simat.
Makes sense.  The voltage is required to calculate the current.
So I'll play with it and see what I can figure out.

simat, 2017-03-04:
As you say the current is calculated by measuring the voltage across the current shunts and converting these to a current. These voltage inputs are defined by the [Current Inputs] section of battery.cfg.

The primary reason for this project was to measure the voltages on each of the individual cells in a lithium battery. Current monitoring was a secondary objective.

SImon

simat, 2017-03-05:
Hi George,
I have been thinking further about this. Currently my software handles a variable number of voltage inputs and two current inputs.

What you want is four current inputs. It wouldn't be too difficult to change the software to take a variable number of current inputs. The number of inputs would be defined in the [Current Inputs] section of the config file.

I am not sure what programming skills you have. Do you want to try to change the code with help from me or would you like me to do it?

Simon

harmongt01, 2017-03-05:
Hi Simon,
Can you help me to understand these current cfg lines.  chP, chN, pga and sps?
[CurrentInputs]
#ibat = 'atod2.readADCDifferential(chP=1, chN=2, pga=2048, sps=250)'
#iin = 'atod2.readADCDifferential(chP=2, chN=3, pga=2048, sps=250)'

I have added one battery voltage monitor point with voltage divider.  The voltage will be at channel=0 using SingleEnded measurement.
I would like to use the second two channels for a Differential reading from a current shunt.
Without the current positions enabled I still get the following error on 'import batteries'.  I probably have something connected wrong but wanted to be sure this cfg should give me a single voltage reading.

>>> import batteries
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "batteries.py", line 32, in <module>
    config['alarms'][i][1] = compile(config['alarms'][i][1], '<string>', 'exec') 
TypeError: expected a readable buffer object

harmongt01, 2017-03-06:
Hi Simon,

Just saw your last post (must not have hit the refresh).  I don't have the programming experience but would like to get a base code working that I can learn from.  And would be willing to pay you for your time to assist.  You can email me at harmongt@live.com.

George

simat, 2017-03-06:
Hi George,

I am happy to help, if we continue to do it through Github it may help others as well.

> Can you help me to understand these current cfg lines. chP, chN, pga and sps?
[CurrentInputs]
#ibat = 'atod2.readADCDifferential(chP=1, chN=2, pga=2048, sps=250)'
#iin = 'atod2.readADCDifferential(chP=2, chN=3, pga=2048, sps=250)'

In this statement
>   self.rawcurrent = eval(config['CurrentInputs']['ibat']) # Battery current in counts

in the module getdata.py my program executes the code you specify in the [CurrentInputs] section of the config file to get the differential voltage reading which equates to the current.

In this case  'atod2.readADCDifferential()' is a function in the Adafruit driver ADS1x15.py. The arguments passed to the function are contained in the brackets. chP=x is the channel number of the positive differential input, chN=y is the negative channel, pga=a sets the amplifier gain of the A-D converter and sps=b is the number of samples per second.

For two current inputs from two 50mV shunts the best setup would be

ibat = 'atod2.readADCDifferential(chP=1, chN=3, pga=256, sps=250)'
iin = 'atod2.readADCDifferential(chP=2, chN=3, pga=256, sps=250)'

pga=256 gives a voltage span of 256mV. It is best to use channel 3 as the common negative channel as it means you can have three differential inputs.

For your voltage input
v1 = 'atod0.readADCSingleEnded(channel=0, pga=2048, sps=250)'

Here is a link to the data manual for the ADS1115, [https://www.google.com.au/url?sa=t&rct=j&q=&esrc=s&source=web&cd=2&ved=0ahUKEwjr3u3P3bvSAhWFFpQKHecRCPgQFgggMAE&url=http%3A%2F%2Fwww.ti.com%2Flit%2Fds%2Fsymlink%2Fads1115.pdf&usg=AFQjCNHQ7BsG5xWhHSCJDDl9uvuSIcA3mQ&sig2=m1bk-fMmHNngutbux6vPcQ](url)

Simon

harmongt01, 2017-03-06:
Hi Simon,

Thank you for the explanation.

updated my config but still see this error:
>>> import batteries
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "batteries.py", line 32, in <module>
    config['alarms'][i][1] = compile(config['alarms'][i][1], '<string>', 'exec') 
TypeError: expected a readable buffer object

George

harmongt01, 2017-03-06:
The link didn't work for me but I did find ADS1115.pdf from Texas instuments.

simat, 2017-03-06:
Hi George,

Sorry my mistake, I have change the code that checks the alarms but haven't updated the default config files. Replace the line in the [alarms] section of the config file with this.

alarm = ['GPIO.setup("P8_10", GPIO.OUT)', 'test=minvolts<2.8 or maxvolts>3.60', 'GPIO.output("P8_10", GPIO.HIGH)', \\
        'test=minvolts>2.9 and maxvolts<3.59', 'GPIO.output("P8_10", GPIO.LOW)']

simat, 2017-03-06:
Other option is to just comment out the alarm.

I have updated the default config files

simat, 2017-03-06:
I have fixed the link to the ADS1115 documentation

harmongt01, 2017-03-06:
Hi Simon,

after replacing the line in the alarms section I get this error:
>>> import batteries
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "batteries.py", line 24, in <module>
    from config import config
  File "config.py", line 31, in <module>
    config = Config()
  File "config.py", line 28, in __init__
    self.config[section][key] = literal_eval(val)
  File "/usr/lib/python2.7/ast.py", line 49, in literal_eval
    node_or_string = parse(node_or_string, mode='eval')
  File "/usr/lib/python2.7/ast.py", line 37, in parse
    return compile(source, filename, mode, PyCF_ONLY_AST)
  File "<unknown>", line 1
    ['GPIO.setup("P8_10",GPIO.OUT)','test=minvolts<2.8 or maxvolts>3.60','GPIO.output("P8_10",GPIO.HIGH)',\'test=minvolts>2.9 and maxvolts<3.59','GPIO.output("P8_10",GPIO.LOW)']
                                                                                                                                                                                ^
SyntaxError: unexpected character after line continuation character

------------------------------
Then when I comment it out, the import batteries works but when I try to run the deamon it throws this error:
Python 2.7.9 (default, Aug 13 2016, 17:56:53) 
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import batteries
>>> batteries.deamon()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "batteries.py", line 41, in deamon
    logsummary = summary.Summary()
  File "summary.py", line 61, in __init__
    if self.summary['yeartodate']['timestamp'][0:4] != printtime[0:4]:
KeyError: 'timestamp'

simat, 2017-03-06:
On the line after the [yeartodate] label in the summary file add this line
timestamp = '20140101000000'

simat, 2017-03-06:
> ['GPIO.setup("P8_10",GPIO.OUT)','test=minvolts<2.8 or maxvolts>3.60','GPIO.output("P8_10",GPIO.HIGH)','test=minvolts>2.9 and maxvolts<3.59','GPIO.output("P8_10",GPIO.LOW)']

I don't think the two lines copied and pasted properly.

If you do git pull you will get the latest updated sample config files and the fixed summary file.

harmongt01, 2017-03-07:
how should I do the git pull?
from the BatteryMonitor directory run this command?  
git clone https://github.com/simat/BatteryMonitor.git

harmongt01, 2017-03-07:
ok.. just git pull
I'm new at this.

harmongt01, 2017-03-07:
Ok.. still moving forward 👍 

getting this error:
>>> import batteries
>>> batteries.deamon()
Error accessing 0x48: Check your I2C address
Error accessing 0x48: Check your I2C address
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "batteries.py", line 49, in deamon
    batdata = Readings()  # initialise batdata after we have valid sys time
  File "getdata.py", line 104, in __init__
    self.getatod()
  File "getdata.py", line 86, in getatod
    self.rawvolts[i+1] = eval(vin[i])/1000 # A to D 1 to 4 in volts
  File "<string>", line 1, in <module>
  File "ADS1x15.py", line 223, in readADCSingleEnded
    val = (result[0] << 8) | (result[1])
TypeError: 'int' object has no attribute '__getitem__'

harmongt01, 2017-03-07:
if no pinouts are connected to assign the ADS address and I have only one ADS1115 then the default should be the 0x48 right?

simat, 2017-03-07:
Do you have pullup resistors on the I2C clock (scl) and data (sda) lines. you need a 10kOhm resistor between these lines and Vdd (3.3V)

simat, 2017-03-07:
Pin 1 (addr) need to be connected to GND

harmongt01, 2017-03-07:
Between the ADS (scl/sda) to BBB (scl/sda) I do not have 10kOhm resistors.. 
Between the ADS VDD and the BBB ADS_VDD I do not have 10kOhm resistor either.  I do not see that in the schematic.  I will add those.  
P9 pin1 BBB should be connected to GND of the battery bank?

Hope I didn't break it....

simat, 2017-03-07:
The pullup resistors need to be between scl and Vdd and sda and Vdd, not between the ADS and BBB.

If your BBB is being powered by a non isolated power supply that is being powered from your battery the GND will be connected via the power supply.

harmongt01, 2017-03-07:
Ok got it.  Will make the changes.  
The BBB is powered by an AC power supply adaptor for now.

harmongt01, 2017-03-07:
Hi Simon,

I will try this tomorrow.  Thank you very much for your help!

-George

simat, 2017-03-07:
The AC power supply will be isolated from the battery so you will need to connect the battery - to P9 pin 1.
I checked my circuit diagram and have realised that the BatteryMonitor.pdf circuit diagram is an old out of date one. I have uploaded a new one.

As you are not monitoring all the individual cell voltages your circuit will be quite simple. Can you upload your circuit so I can see what you have come up with.

Simon

harmongt01, 2017-03-07:
[image]
Here's a photo but it's really hard to see what's connected to where.
I still have the same issue so I will pull out the connections and start over.  Do you think I should see some kind of output if I have no voltage or shunts connected?

simat, 2017-03-07:
You shouldn't be getting any errors and be getting some kind of output without any inputs connected.

I see you have two ADS1115s connected. Is each one on a separate address? The one with address 0x48 should have pin 1 (ADDR) connected to GND and the second one (address 0x49) should be at Vdd (3.3V)

To get some sort of output form the ADS1115s all you need to connect is 
Pin 1 = GND
Pin 3 = GND
Pin 8 = Vdd (3.3V)
Pin 9 = SDA Beagle P9-20
Pin 10 = SCL Beagle P9-19

Check these connections first and if you cant get anywhere I suggest you strip your circuit down to the bare minimum and see if it works.

Simon

harmongt01, 2017-03-07:
Yes I have two ADS1115.  Right now only the one on the right is connected to the BBB.  The jumper from ADDR is connected to GND.

I will have to strip it down to the minimum as you suggested.

simat, 2017-03-07:
There is a linux package i2c-tools that has a program called i2c-detect
If you run i2c-detect -r 1 you should get the following with one ADS1115

> WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will probe file /dev/i2c-1 using read byte commands.
I will probe address range 0x03-0x77.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- 48 -- -- -- -- -- -- -- 
50: -- -- -- -- UU UU UU UU -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --

harmongt01, 2017-03-08:
I have it stripped down.  
Trying to run the detect program:

root@beaglebone:~# apt-get install i2c-tools
Reading package lists... Done
Building dependency tree       
Reading state information... Done
i2c-tools is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 21 not upgraded.
root@beaglebone:~# i2c-detect -r 1
bash: i2c-detect: command not found
root@beaglebone:~#

harmongt01, 2017-03-08:
not sure why it has the strike through on those lines.  Not copying over correctly I guess.

harmongt01, 2017-03-08:
I tried this:
root@beaglebone:~# i2cdetect -y 1
Error: Can't use SMBus Quick Write command on this bus

simat, 2017-03-08:
I don't understand why i2cdetect -r 1 doesn't work
Here is the response I get to trying different options with i2cdetect

> simat@beaglebone:~$ i2cdetect
Error: No i2c-bus specified!
Usage: i2cdetect [-y] [-a] [-q|-r] I2CBUS [FIRST LAST]
       i2cdetect -F I2CBUS
       i2cdetect -l
  I2CBUS is an integer or an I2C bus name
  If provided, FIRST and LAST limit the probing range.

> simat@beaglebone:~$ i2cdetect -l
i2c-0	i2c       	OMAP I2C adapter                	I2C adapter
i2c-1	i2c       	OMAP I2C adapter                	I2C adapter

> simat@beaglebone:~/BatteryMonitor$ i2cdetect -r 1
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will probe file /dev/i2c-1 using read byte commands.
I will probe address range 0x03-0x77.
Continue? [Y/n] y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- 48 49 4a -- -- -- -- -- 
50: -- -- -- -- UU UU UU UU -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --    

> simat@beaglebone:~$ i2cdetect -y 1
Error: Can't use SMBus Quick Write command on this bus

I would try the i2cdetect -l first to see if the i2c interfaces are working then the i2cdetect -r 1
                     
Hope this is useful, Simon

simat, 2017-03-08:
Hi George,
Had a bit of spare time and looked at your circuit in more detail.
The Adafruit boards have the 10k pullups for SCL and SDA and a pull down for ADDR so you could leave ADDR disconnected and have no external pullups for the circuit to work at address 0x48.

Would you like me to look at the circuit diagram of your current sensing and battery voltage divider circuits?

SImon

harmongt01, 2017-03-08:
HI Simon,

I ran those commands:
root@beaglebone:~# i2cdetect -l
i2c-0   i2c             OMAP I2C adapter                        I2C adapter
i2c-1   i2c             OMAP I2C adapter                        I2C adapter
i2c-2   i2c             OMAP I2C adapter                        I2C adapter

root@beaglebone:~# i2cdetect -r 1
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will probe file /dev/i2c-1 using read byte commands.
I will probe address range 0x03-0x77.
Continue? [Y/n] Y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                     

I have the ADS down to only the basic connections.  VDD, GND, SCL and SDA.  I removed the jumper between GND and ADDR so the address should still be 0x48.

harmongt01, 2017-03-08:
I think my problem is down to the BBB not seeing the ADS boards.  I tried both boards one at a time and same thing.  Nothing found with i2cdetect.  

Possible problems as I see it:
1.  Either both ADS boards are bad 
2.  something is wrong with the SCL and SDA ports at the BBB
3.  SCL and SDA ports not enabled.  I read these should be enabled by default.
4.  driver for ADS boards not installed?

harmongt01, 2017-03-08:
Hi Simon,

I don’t have any diagrams for my current sensing or voltage divider.  The voltage at the divider output point measured 2.6VDC when I first put it in.

I have two 100A 75mV shunts in place.  These are in series with the negative sides of my input from solar panels and input from small wind generators.  Wind and solar on different shunts.

I have a feeling something is not right with my BBB SCL and SDA but I don’t know how to check if the ports are ok.

Thanks,
George

simat, 2017-03-08:
> root@beaglebone:~# i2cdetect -l
i2c-0 i2c OMAP I2C adapter I2C adapter
i2c-1 i2c OMAP I2C adapter I2C adapter
i2c-2 i2c OMAP I2C adapter I2C adapter

I see you have the new BBB with WiFi. Interestingly it has three i2c devices listed where my old BBB has two.

Try i2cdetect -r 2 and then i2cdetect -r 0

harmongt01, 2017-03-09:
I have different results with those commands, looks more promising:

root@beaglebone:~# itcdetect -r 0
bash: itcdetect: command not found
root@beaglebone:~# i2cdetect -r 2
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will probe file /dev/i2c-2 using read byte commands.
I will probe address range 0x03-0x77.
Continue? [Y/n] Y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- 48 -- -- -- -- -- -- -- 
50: -- -- -- -- UU UU UU UU -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
root@beaglebone:~# i2cdetect -r 0
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will probe file /dev/i2c-0 using read byte commands.
I will probe address range 0x03-0x77.
Continue? [Y/n] Y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- UU -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- UU -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: UU -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: UU -- -- -- -- -- -- --

harmongt01, 2017-03-09:
How can you tell from that output there is an onboard Wifi?  I wouldn't have thought that would be a factor but makes sense.  You are good at this!

simat, 2017-03-09:
Great!!! OK this means that your A-D is on bus 2 rather than 1.

If you change the line atod0 in the [AtoDs] section of battery.cfg to 

atod0 = 'AtoD.ADS1x15(busnum=2,address=0x48,ic=0x01, debug=True)'

my software should hopefully work!

harmongt01, 2017-03-09:
Thanks Simon.  I updated the config.  Need to rewire the connections.. will send status shortly.

simat, 2017-03-09:
> How can you tell from that output there is an onboard Wifi? I wouldn't have thought that would be a factor but makes sense. You are good at this!

I could see from your photo.

My earlier version of the board has an error where i2c bus 1 is connected to pins 19 and 20 even though the pins are called I2C2-SCL and I2C2-SDA. I now remember this caused me some grief when I first designed this. Looks like the fault has been fixed with your version of the BBB.

harmongt01, 2017-03-09:
Ah yes. the picture shows the wifi wires.

I just connected things back up and when trying to start the deamon it threw this error:

root@beaglebone:~/Adafruit_Python_ADS1x15/BatteryMonitor# python
Python 2.7.9 (default, Aug 13 2016, 17:56:53) 
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import batteries
>>> batteries.deamon()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "batteries.py", line 49, in deamon
    batdata = Readings()  # initialise batdata after we have valid sys time
  File "getdata.py", line 104, in __init__
    self.getatod()
  File "getdata.py", line 87, in getatod
    self.rawcurrent = eval(config['CurrentInputs']['ibat']) # Battery current in counts
  File "<string>", line 1, in <module>
NameError: name 'atod2' is not defined
[image]

harmongt01, 2017-03-09:
the voltage out going to A0 was 2.58VDC when I connected it.

simat, 2017-03-09:
ops my mistake, can you change atod2 in the lines in the [CurrentInputs] section to atod0, this corresponds to the AtoD address you defined in the [AtoDs] section

If your voltage input is 2.58V you will need to change pga=2048 to pga=4096 in the [VoltageInput] section

harmongt01, 2017-03-09:
Next error:

root@beaglebone:~/Adafruit_Python_ADS1x15/BatteryMonitor# python
Python 2.7.9 (default, Aug 13 2016, 17:56:53) 
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import batteries
>>> batteries.deamon()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "batteries.py", line 49, in deamon
    batdata = Readings()  # initialise batdata after we have valid sys time
  File "getdata.py", line 104, in __init__
    self.getatod()
  File "getdata.py", line 98, in getatod
    self.uncalvolts[i] = self.rawvolts[i]*config['calibrate']['batvgain'] # A/D to battery volts
KeyError: 'batvgain'
>>>

harmongt01, 2017-03-09:
Just checked the reading at the voltage divider:  2.68VDC
Voltage at the battery bank is 13.23VDC.

simat, 2017-03-09:
Add this line in [calibrate] section

batvgain = 4.94 # resistor divider gain 13.23/2.68

I will be away from the computer for a couple of hours. I am in Western Australia. The time here is 12:01pm

harmongt01, 2017-03-09:
I'd love to explore the northwestern side of Australia some day... on my bucket list

harmongt01, 2017-03-09:
>>> import batteries
>>> batteries.deamon()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "batteries.py", line 49, in deamon
    batdata = Readings()  # initialise batdata after we have valid sys time
  File "getdata.py", line 104, in __init__
    self.getatod()
  File "getdata.py", line 91, in getatod
    +config['calibrate']['pcurrent'] # current in amp  
KeyError: 'ibatgain'
>>>

simat, 2017-03-09:
I think there is an error in the [calibrate] section of the config file. Can you upload it to me again
Thanks Simon

[28 more replies]

closed 2017-03-12 by simat