simat/BatteryMonitor #3: Voltage Caluclation

harmongt01, 2017-03-13:
Hi Simon,

After running a couple tests on the voltage reading, I have the following results:

Using a voltmeter I took measurements of the battery voltage and the output of the voltage divider.  

Battery Bank Voltage = 12.86VDC (voltmeter reading)
Vout = 2.288 (voltmeter reading at voltage divider)
BatteryMonitor calculation = 12.45

----------Then charged the battery bank up by 1V ---------------

Battery Bank Voltage = 13.86VDC (voltmeter reading)
Vout = 2.305 (voltmeter reading at voltage divider)
BatteryMonitor calculation = 12.523

I'm not sure where to adjust the calculation to make the readings align.  I tried to get the bank voltage as close to an even 1V apart as I could for this sample.

Attached is the current config.

[battery.txt](https://github.com/simat/BatteryMonitor/files/839878/battery.txt)

Thanks,
George

simat, 2017-03-14:
Hi George,
That ratio of Vin to Vout of the resistor divider should stay constant. There must be something wrong.

Using my circuit diagram you should be taking the voltage reading across R1. What is the make and model of your voltmeter?

Simon

Simon

harmongt01, 2017-03-15:
Hi Simon,

Certainly the mistake is with my voltage divider.
I made a change.  Voltmeter is a Craftsman, I'd have to pull the cover to check the model but it's been very reliable.  Had it for years, really don't think it's the problem.  My skills are to question.

Here are the results of this last test after a change to the divider:

Battery Bank Voltage = 13.95
Vout = 3.03
BatteryMonitor calculation = 13.062
-----------------------------------------
Battery Bank Voltage = 12.95
Vout = 2.8
BatteryMonitor calculation = 13.02

Percentage of change is much closer now between Vout and Battery Bank:
Battery bank voltage.....12.95v to 13.95v difference is ~ .928315%
Vout........................................2.8v     to 3.03v difference is ~ .924092%

BattMon...........................13.02v to 13.062v difference is ~ .996785%

I assume the difference between Vout and Battery voltage should be proportional.  Is the difference I have now something I can correct with a calculation in your program?  The battery bank should always stay within about 1V between low and full charge using the charge controller as I understand??  So with the given battery charge limit of +/-1V can I compensate for the difference with accuracy?

Thanks,
George

harmongt01, 2017-03-15:
not sure why some lines printed in bold...

simat, 2017-03-15:
Hi George,
batvgain in the [calibrate] section of the config file sets the battery monitor output. It should be set to the ratio Vbattery/Vout. 

At Vbat=12.95 the ratio is 4.625, at Vbat=13.95 the ratio is 4.604 so I would set the average of the two which is 4.6145

I am puzzled that you are getting a 0.5% discrepancy between the ratio at the different voltages. What value resistors have you chosen for R1 and R2.

I would expect the output from the battery monitor to match the battery voltage better than it is.

You do have the battery voltage input on ain0 (pin 4)?

Is your circuit the same as mine?

Is the battery - connected to Vss (GND pin on the A to D)?

Simon

harmongt01, 2017-03-16:
Hi Simon,

I believe I solved my problem with the voltage difference.  Using a capacitor rated for higher voltage may have been the problem.  You can close this issue.  Here is a photo with some details of my wiring as it is now.  Voltage readings look great.

[image]

Thanks George

closed 2017-03-17 by harmongt01