simat/BatteryMonitor #5: What is the VDD_ADC?

harmongt01, 2017-03-19:
Hi Simon,
 
BTW, no rush to answer my questions.  I'm making great progress on this project thanks to your program and assistance.

I don't understand how the VDD_ADC should be connected.  From the BB documentation it mentions this is a 1.8V reference.  I can't tell from your schematic how you are using it?

Thanks in advance,
George

simat, 2017-03-19:
Hi George,

Sorry, I have been away.

You are right. For some reason I must have thought that VDD_ADC was 3.3 volts. I have measured it and as you say it is really the 1.8 volts ADC voltage reference.  

The ADC1115s I have been using have been working happily for a couple of years with a supply voltage of 1.8 volts!

I will change the circuit diagrams to use the +3.3V on P9 pins 3,4

Thanks, Simon

harmongt01, 2017-03-20:
Hi Simon,

Ok, so that makes sense.  Since the pin is named VDD_ADC it was a little confusing.

Thanks,
George

closed 2017-03-21 by simat