simat/BatteryMonitor #6: Additional Current Readings

harmongt01, 2017-03-20:
Hi Simon,

Can you give me some instruction or assistance on how to add two or three more current readings?

Thanks,
George

simat, 2017-03-21:
Hi George,
Currently my software is hardwired to read two current inputs, battery current and solar input and can calculate a third, the total output current by subtracting the battery current from the solar input.

You could define a new current input in the [CurrentInputs] section of the config file, something like,
iin2 = 'atod2.readADCDifferential(chP=2, chN=3, pga=256, sps=250)'
but you would have to then write some code to get this input filtered and put into the log files.

I was thinking about changing my code so it would handle current inputs like it does voltage inputs which would automatically log a variable number of current inputs.

Any thoughts?

How many current inputs were you thinking of monitoring?

Simon

harmongt01, 2017-03-21:
Hi Simon,

In my application, I'd like to measure current inputs from multiple (small) wind generators and inputs from a couple different solar arrays (eventually multiple).  There will be more than one load as well that I'd like to measure the current (output).  All measurement points will be 12VDC.

So changing your code to measure current inputs the way it is voltage would be ideal in my scenario.  

If you could help me with that It would be perfect.  I'm already very pleased with the results of your code as it is.  I can see what's going on now.  I can see I have some things to work out as well, thanks to the data and trends.

Anything I can do to help?  Not urgent of course.

Thanks,
George

simat, 2017-03-22:
Hi George,

I am intrigued to know why you are looking the power from/to each source and output separately. Is it some sort of device efficiency testing?

OK, I will make the changes to my code over the next few weeks. The changes will show up as extra comma separated entries in the log file and in the 'amp' entry in the summary files.

What you could do is maybe add a page to and/or edit my WIKI detailing how to use this project to log information from many different power sources/loads and monitoring just a whole battery rather than the individual cells.

Simon

harmongt01, 2017-03-22:
Hi Simon,

There are a few reasons for monitoring the multiple inputs separately.  The main reason is to study what power I'm actually dealing with.  Without historical data it's been more of a black box.  The other reason is so it will be more obvious if something is not working as it should.  By monitoring each input separately I should be able to see the comparison of one input or output vs another, such as comparing two wind generators.

I'd be glad to add to the wiki, good idea.

Thanks,
George

harmongt01, 2017-03-22:
Greetings again Simon,

I'm starting to suspect my multimeter may not be accurate (as you already questioned).  It's a Craftsman 82027.  I've had for many years but never used it in the way I'm using it now, but it's the only true reference tool you can use right?  So if I were to upgrade what would you recommend?  I know the Fluke brand is good.  The model 115 looks interesting but seems a little expensive for a multimeter to me.

When using my voltmeter, the voltage readings are very close (I think).  However, when measuring the current (bypassing the shunt), the readings seem off from what I'd expect.  

Thanks,
George

simat, 2017-03-23:
Hi George,
Ah, trying to read current accurately, now there is a can of worms.

The big problem is that although the voltage is kept reasonably constant by the battery the current being supplied by the wind turbines, solar panels and drawn by the loads is varying with time. This variation occurs 
over a frequency range of kHz down to less than 1Hz. Different multimeters average out this variation in different ways so will give you different results.

The best way to measure the current is to measure the voltage across the shunt. This voltage has to be filtered to smooth out the current variations.

If you look at my circuit for the current inputs the two 100k resistors and the 0.1uF capacitor plus software averaging do the smoothing.

If you want to do the same thing with your multimeter, use one 10k resistor in series with a 10uF capacitor and measure the voltage across the capacitor.

I used to use Fluke multimeters at work but I can't justify the expense for the amount of use I have for a multimeter nowadays. I now have a Uni-T 61E which works very well. If you want to check the voltage accuracy of your multimeter something like this is useful. [http://shop.voltagestandard.com/product.sc?productId=1&categoryId=1](url)

Simon

harmongt01, 2017-03-24:
Excellent Simon.

Thank you.  I had to order some 10uF capacitors.  My inventory is slim.

simat, 2017-03-24:
Hi George,

I have made the changes so you can have as many current sources as you want.

I am in a bit of a hurry so can't give much of an explanation at the moment

Just look at the new file battery1cell.cfg and summary.1cell to get an idea what I have done.

Do a git pull to get the changes

Simon

harmongt01, 2017-04-02:
Hi Simon,

It looks really great.  I had it running then it today something keeps triggering this error:

>>> batteries.deamon()
13.827 13.83 8.827 0.0 -1.2 9034.24 9042.93
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "batteries.py", line 82, in deamon
    and batdata.batcurrentav < config['battery']['ireset']:  # reset SOC counter?
AttributeError: Readings instance has no attribute 'batcurrentav'

Thanks,
George

harmongt01, 2017-04-02:
from the error it looks like you added batcurrentav as part of the new code perhaps but didn't define it.  I added this line in getdata.py
  batcurrentav = [ 0.0 for i in range(len(iin))]

Now it is running but I'm not certain I defined this attribute as you would have intended.

George

simat, 2017-04-02:
Hi George,
Thanks for picking that bug up. 
The problem was that the original code only had the hardwired current inputs batcurrent and incurrent. This of course has been changed to have the variable number of current inputs in the list current[].

My battery monitor had stopped a few days ago and I hadn't got around to finding out what the problem was.

The piece of code where the error occurred checks to see if the battery voltage has reached the fully charged voltage config[battery][vreset] and that the current has dropped below config[battery][ireset]. If these conditions are true the battery is fully charged so the SOC counter gets reset.

To fix this you have to change line 82 in batteries.py to the following and delete the modification you made in getdata.py.

          and batdata.currentav[0] < config['battery']['ireset']:  # reset SOC counter?

The other way to change the code is to revert back to the original getdata.py with the command git checkout getdata.py and then do a git pull.

Simon

harmongt01, 2017-04-03:
Ah... perfect!

simat, 2017-04-05:
Hi George,
I have uploaded a new program called caligain.py to calibrate the current inputs. To run just type 'python caligain.py' and follow the instructions. There is already a  program to calibrate the voltages, type 'python calall.py' to do this. I am thinking of completely changing the way the voltages are calculated and displayed. You will have to recalibrate the voltage readings if I do this.

Hope things are progressing and you are not too bogged down with work.

Simon

harmongt01, 2017-04-06:
Hi Simon,

Work is busy, which is good.  Working overnight isn't my favorite especially when changes don't go well.

If you want to change your program I'm good with that.  BTW how can I get the additional current readings to write to the log?  I still haven't received the additional current shunts yet.  I think they'll be here sometime next week.

Thanks,
George

simat, 2017-04-07:
The additional current readings will appear automatically. Just have to define the inputs in the config file

harmongt01, 2017-04-07:
It doesn't like the channels I added to the third current input line in my config file.  I should have channels 0-3 again on the second ADS, atod1?

[CurrentInputs]
iin1 = 'atod0.readADCDifferential(chP=1, chN=3, pga=256, sps=250)'
iin2 = 'atod0.readADCDifferential(chP=2, chN=3, pga=256, sps=250)'
iin3 = 'atod1.readADCDifferential(chP=1, chN=2, pga=256, sps=250)'

simat, 2017-04-07:
In the [AtoDs] section have you defined atod1?

simat, 2017-04-07:
The problem is the ADS1115 can only use channels 0-1,0-3,1-3,2-3 in differential mode.

I would put all the current inputs on atod0 with channel 3 as the common and move the voltage input to atod1

harmongt01, 2017-04-10:
Got it.  Ok yeah good idea.
I've had my BBB down for a few days trying to get a small touchscreen installed.  I probably don't need it but just wanted it for displaying data and access to the box while I'm local to it.  Not as simple as installing a driver apparently.  It came with an image for BBB or Rasberri Pi to use it.  

George

simat, 2017-04-13:
Hi George,
I have done a bit of a reorganisation of the way the current, power and energy is stored in the summary files. The changes are reflected in the new summary file template summary.1cell
If you look at the file you will see the following in each of the sections.

ioutmax  = [0.0, 0.0, 0.0, 0.0]
kwoutmax = [0.0, 0.0, 0.0, 0.0]
kwhout   = [0.0, 0.0, 0.0, 0.0]
iinmax   = [0.0, 0.0, 0.0, 0.0]
kwinmax  = [0.0, 0.0, 0.0, 0.0]
kwhin    = [0.0, 0.0, 0,0, 0.0]

ioutmax is the maximum output current in amps, I define output as positive
kwoutmax is the maximum output power in kW
kwhout is the total energy out in kWh
The other three are the corresponding entries for current in the opposite (negative) direction.
Simon

simat, 2017-04-13:
How did you go getting your display to work?

harmongt01, 2017-04-17:
Hi Simon,
Thanks for the update.  I will pull it and check.  I have not been successful getting the display working yet.  

George

harmongt01, 2017-04-18:
Simon,
I updated your changes.  Also added two additional shunt readings (total of four now) to the config file.  The main program is running fine and displaying all four.  However, I tried to run the 'python caligain.py' module and it is failing with the following error:

root@beaglebone:~/Adafruit_Python_ADS1x15/BatteryMonitor# python caligain.py
Usage: At > prompt enter current input channel number to change then [Enter] key
       Pressing just the [Enter] key will display the latest current readings, type ^C to exit

       Make sure the Battery Monitoring software is running otherwise the summary file will not be current
       This program uses the data from the log file generated by the Battery Monitoring software
       Make sure that the Battery Monitoring software is logging to a real file and not /dev/null
       This program averages the data from the log file for the last 100 seconds.
       To make sure that the calibartion is accurate make sure the current on the input being calibrated is not varying
Traceback (most recent call last):
  File "caligain.py", line 107, in <module>
    main()
  File "caligain.py", line 86, in main
    geti()
  File "caligain.py", line 67, in geti
    iin=literal_eval(summaryfile.get('current','currentmax'))
  File "/usr/lib/python2.7/ConfigParser.py", line 618, in get
    raise NoOptionError(option, section)
ConfigParser.NoOptionError: No option 'currentmax' in section: 'current'

I looked at the script briefly but I'm not sure.  Maybe something to do with the additional current readings?  Attached my running config file.
[battery.cfg.txt](https://github.com/simat/BatteryMonitor/files/927162/battery.cfg.txt)

Thanks,
George

simat, 2017-04-18:
Hi George,
Sorry about that, caligain needs to be modified for the new current readings.
'currentmax' needs to be changed to 'ioutmax'

Do a git pull to hopefully fix the problem

Simon

harmongt01, 2017-04-18:
Hi Simon,
That was it... perfect!

If I'm seeing two of the four current readings with a negative value.  Would that be a setting I should adjust?  All current shunts are on the negative side of the battery bank but I would think the differential measurements would be either all positive or all negative.

3.17 13.22 8.170 -0.00 0.000 -0.00 0.000 9997.18 9997.61

Thanks,
George

simat, 2017-04-18:
Hi George,  the polarity of the current reading is dependent on which direction the current is flowing through the shunt resistor. 

Below is a diagram of how I have organised my current shunts and my current input configuration
                                     2
                   1       |----/\/\/\/\----Solar Controller
Battery ---/\/\/\/\---|
                            |-------------- Loads
                                     3
[CurrentInputs]
iin1 = 'atod2.readADCDifferential(chP=0, chN=1, pga=256, sps=250)'
iin2 = 'atod2.readADCDifferential(chP=2, chN=3, pga=256, sps=250)'
iin3 = 'self.current[0] - self.current[1]'

Current shunt 1 measures the battery current and I have set it up so that current going into the battery is negative and coming out of the battery is positive.

Current shunt 2 measures the current from my solar panels, charging current from the panels to the battery is negative and at current drawn by the solar controller at night is positive.

There is no current shunt 3 measuring the current going to the loads. It is calculated by subtracting the current coming from the solar controller from the battery current. This current reading is always positive.

When you use caligain just type in - before the current reading if you want the reading to be minus.

Simon

harmongt01, 2017-04-19:
Hi Simon,

I've been working a bit late here but wanted to ask about your last post.  

[CurrentInputs]
iin1 = 'atod2.readADCDifferential(chP=0, chN=1, pga=256, sps=250)'
iin2 = 'atod2.readADCDifferential(chP=2, chN=3, pga=256, sps=250)'
iin3 = 'self.current[0] - self.current[1]'

How would you filter this kind of configuration with your schematic?  I was using something different according to your earlier design..  This one thorws me off just a bit.  I need another filter schematic to work form even if it's just on paper.  

Gracias Amigo,

I been up all night with work.  Hope you have a great day and sleep well my friend.

Much thanks for all you do, 
George

simat, 2017-04-20:
Hi George,
With reference to the [ BatteryMonitor/BMSsingle.pdf ](url) circuit diagram the shunt that I have labeled as 1 a couple of posts ago is the shunt called SHUNT between the battery minus terminal and the second shunt called SHUNT1 which is shunt 2 in the earlier post.

SHUNT is connected to the ADS1115 ain1 via the 100k resistor R42 and ain3 via R43 which with capacitor C6 make up the filter.

SHUNT1 is connected to the ADS1115 ain2 via the 100k resistor R42 and ain3 via R43.

Simon

harmongt01, 2017-04-24:
HI Simon,
I have the filter set up like the pdf shows.  I was playing around with this over the weekend to try to adjust the gains for the current readings.  For some reason, however, I cannot get more than one current reading to work.  It seems like the filter is ok.  So for example, I can get a reading on AIN1 when AIN3 is the common.  No reading will show for AIN2.  If I move the pin that is in AIN2 and swap it with AIN1, then it gives a reading in AIN1.  Seems like an issue with the ADS1115?  Maybe I broke something... strange problem.  I have more than one ADS1115.  Seems like I have the same issue for each.

George

simat, 2017-04-24:
Hi George, does sound strange, what are the [current input] config lines?

harmongt01, 2017-04-24:
At the moment I'm using this:

[CurrentInputs]
iin1 = 'atod1.readADCDifferential(chP=0, chN=1, pga=256, sps=250)'
iin2 = 'atod1.readADCDifferential(chP=0, chN=3, pga=256, sps=250)'
iin3 = 'atod2.readADCDifferential(chP=0, chN=1, pga=256, sps=250)'
iin4 = 'atod2.readADCDifferential(chP=0, chN=3, pga=256, sps=250)'

simat, 2017-04-25:
Hi George,
If you were swapping between ain 1-3 and ain 2-3 try this for your configuration.
[CurrentInputs]
iin1 = 'atod1.readADCDifferential(chP=1, chN=3, pga=256, sps=250)'
iin2 = 'atod1.readADCDifferential(chP=2, chN=3, pga=256, sps=250)'

Your configuration should work if you have ain0 as the common and have the other side of the shunts on ain1 and ain3.
Simon

harmongt01, 2017-04-25:
Yes I tried it multiple ways but thinking I may have done something to the ADS115's or maybe I have a wire not connected well.  I will continue to play with it this evening.  I was getting a single current measurement and somehow I've succeeded to break that as well 👍

simat, 2017-04-25:
It sounds like two steps forward, one step back.

I just had a thought that if you entered 0 as a current when using caligain that it would set the current multiplier in the battery config file to zero so you would always zero current.

In the [calibrate] section of the config file you should have something like this
currentgain = [0.010417, 0.005208, 0.0010417, 0.0010417]
currentoffset = [-0.0, 0.0, 0.0, 0.0]

If any of the currentgain entries are 0.0 you will get a zero reading. To check the raw data from the ADS1115 you can set the currentgain entries to 1.

Another check you could make with the ain2 input is to set it up as a single ended rather than a differential input with a config something like
iin2 = 'atod0.readADCSingleEnded(channel=2, pga=4096, sps=250)' 
and see if you get any readings.

harmongt01, 2017-04-26:
Hi Simon,

I bet that is it.  In my troubleshooting efforts I did manually set the currentgain to all 0.0.  I see when they are set to 0.0 I get a float division by zero error when executing caligain.py.  Now I've adjusted those manually to something like your example I think it's working.  This evening I will have to do some more tests.

Thanks,
George

harmongt01, 2017-04-27:
Hi Simon,

That was the main problem.  I also had a bad ADS1115.  Everything is perfect now!  

I ran the battery bank down to ~80% and used a charger set at 2A to set the caligain measurments on each shunt.  Worked very well.

We can probably close this issue.

Thanks,
George

simat, 2017-04-28:
Hi George,
Great to hear you have got everything working, well done!

When you are calibrating the current inputs, you will get a much better result if the current used to calibrate is close to or at the maximum input current. If your maximum current input is say 100A you should ideally calibrate with a current of around 100A.

If you are unable to calibrate the current inputs with a current that you can accurately measure that is close to the maximum operating current you might get better accuracy by calculating the gain using the shunt measurement parameters.

The gain can be calculated using the formula GAIN=I@ratedvoltage/(V@ratedcurrent*256) where V is in mV. So if we have a 100A 50mV shunt the GAIN=100/(50*256) which equals 0.0078125.

BTW did you see the comment I made about generating graphs using Gnuplot it the other open issues thread.
Simon

harmongt01, 2017-04-29:
Hi Simon,
Ok wow I didn't know that.  100A would be quite a bit of current.  I thought that if you use a 100A shunt then the power should never really go above ~75%.  

I am using a battery charger to send the current through each shunt seperately (while the others are disconnected).  It can charge at 2A, 10A, 40A, 100A or 200A.  Schumacker SE 4020 charger.  It's probably not ideal since it can sense the level of the battery charge and auto adjust as it approaches fully charged.

Am not sure how to deliver close to 100A.  Seems like it could be dangerous.

I am looking at that post about the Gnuplot now.  I want to play with that.  It looks great.

Thanks,
George

harmongt01, 2017-05-12:
Hi Simon,
I needed to rebuild my BBB after some changes.  I'm stuck again on something, possibly related to a difference of directory structure between Adafruit and BatteryMonitor software locations.

I am getting the below error about a config file.  I am not sure where it is looking for the config module:

root@beaglebone:~/SD# python
Python 2.7.9 (default, Aug 13 2016, 17:56:53) 
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import batteries
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "batteries.py", line 24, in <module>
    from config import config
ImportError: No module named config

Thanks,
George

simat, 2017-05-13:
Hi George,
Looks like different versions have got mixed up. I would create a new clean directory and then do a git clone from github.

You need to use the version of the Adafruit driver that is part of my package as I have patched it to work with multiple I2C buses. I should try to get my changes to their driver incorporated in their version.

Simon

closed 2017-05-13 by simat