simat/BatteryMonitor #7: Timestamp

harmongt01, 2017-03-25:
Hi Simon,
I edited the timestamp format in both batteries.py and summary.py.
Not sure what this error means:

>>> batteries.deamon()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "batteries.py", line 44, in deamon
    while int(printtime) <= int(summary['current']['timestamp']):
ValueError: invalid literal for int() with base 10: '03/25/2017 13:50:07 '

harmongt01, 2017-03-25:
[batteries.py.txt](https://github.com/simat/BatteryMonitor/files/870412/batteries.py.txt)
[summary.py.txt](https://github.com/simat/BatteryMonitor/files/870413/summary.py.txt)

It's strange.  I put back the original files I had backed up but still get the same error.

harmongt01, 2017-03-25:
After putting back the original files, the error is not the same. The format of the date is different in the error message. 
>>> batteries.deamon()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "batteries.py", line 44, in deamon
    while int(printtime) <= int(summary['current']['timestamp']):
ValueError: invalid literal for int() with base 10: 'Mar 24 2017 20:01:09

simat, 2017-03-26:
Hi George,
When you say you changed the timestamp format, do you mean you changed it in the summary file?

simat, 2017-03-26:
Rather than manually keeping different versions of the files you are editing you can get Git to do it.

If you want to go off on a sidetrack you can do what is known as fork with git, play around with the fork and then remerge the fork back into the main branch when you have everything working. You can also edit files in a Git directory and then use a Git command to revert back to the original.

Here is a tutorial on using Git [https://www.atlassian.com/git/tutorials/setting-up-a-repository](url). There are many others if you don't like this one.

harmongt01, 2017-03-26:
Ok Simon,  yes I need to learn it.

I cloned what is currently in the main repository and placed my battery.cfg in the directory.  Now I'm getting this error below.  Not sure what I did wrong. At the moment I should have exactly what you have except for the cfg file.

>>> batteries.deamon()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "batteries.py", line 41, in deamon
    logsummary = summary.Summary()
  File "summary.py", line 55, in __init__
    if self.summary['hour']['timestamp'][0:10] != printtime[0:10]:
KeyError: 'hour'

I was going for a timestamp format of '%m/%d/%Y %H:%M:%S' but ran into trouble lol.

Also I forked another clone into a new repository to start working from when I can get it back to a working directory again.

Thanks,
George

simat, 2017-03-26:
The key error implies to me that the [hour] section header is missing from the summary file. 

Did you want that date format in the summary and log files?

One issue with the date format you are looking at and one of the reasons I use the one I do is that if you do a sort on the timestamp it will sort into chronological order.

Simon

harmongt01, 2017-03-26:
Hi Simon,
Sorting in chronological order is probably what I want.  I was trying to play with the graphing display so the date would be easier to read.  I did try replacing the summary file with the summary.1cell from the master repository.    

[summary.txt](https://github.com/simat/BatteryMonitor/files/870545/summary.txt)

I'm sure I'm missing something simple.  Not understanding how the summary file is used really.

Thanks,
George

simat, 2017-03-26:
Another thought, check the [files] section of your battery config file to make sure that the path to the summary file is correct.

The summary file is where I summarise all the battery stats. [current] are the current battery stats, [hour] are the hourly stats and so on. Look at the "Sample Output" wiki page for more information. The sample file is updated every minute. Each update merges the data from the last minute into all the summaries.

Simon

harmongt01, 2017-03-26:
I understand.  I believe the summary file has the headers correct.  
[battery.cfg.txt](https://github.com/simat/BatteryMonitor/files/871060/battery.cfg.txt)

I don't understand why it still fails if I delete the entire BatteryMonitor directory and clone it again.  If I place the backed up battery.cfg it should work again as it did before I touched the two .py files.

George

harmongt01, 2017-03-26:
Yeah I know I'm all over the place now.  I replaced the summary file and now getting this error.  I have no idea what I'm doing apparently.... I should just clean everything out and start from scratch probably.

>>> batteries.deamon()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "batteries.py", line 49, in deamon
    batdata = Readings()  # initialise batdata after we have valid sys time
  File "getdata.py", line 115, in __init__
    self.getatod()
  File "getdata.py", line 103, in getatod
    self.current[i] = eval(iin[i])*config['calibrate']['currentgain'][i] \
KeyError: 'currentgain'

-George

simat, 2017-03-26:
Hi George,
I have added some new config variables to the config file.
Add the following lines to the [calibrate] section of battery.cfg

# count of 32767=256mV, so 1mV=128 counts
# so gain=current@shuntvoltage/128*shuntvoltage
currentgain = [0.010417, 0.005208, 0.0010417]
currentoffset = [-0.0, 0.0, 0.0]

Simon

harmongt01, 2017-03-27:
Simon,

Added the lines and... It's alive!!

13.064 13.06 8.064 0.0 -0.0 10000.0 10000.01
13.113 13.11 8.113 0.0 -0.0 10000.0 10000.01
13.112 13.11 8.112 0.0 -0.0 10000.0 10000.01
13.131 13.13 8.131 0.0 -0.0 10000.0 10000.01

-George

closed 2017-04-04 by harmongt01