simat/BatteryMonitor #8: Missing graphs in wiki page "Wind and Solar Project"

simat, 2017-03-26:
Hi George,

Great to see your wiki page!

The graphs you mention don't show up when I look at the page.

Simon

harmongt01, 2017-03-26:
I am working on getting those images inserted.

harmongt01, 2017-03-27:
Hi Simon,

Let me know if you can see the graphs now.  If there is a better way to display them I'm open to suggestions.

Thanks,
George

harmongt01, 2017-03-27:
The graphs you have are from the web server on your BB, so realtime ?

simat, 2017-03-27:
I see what you mean about displaying the datestamp in your graphs. What program are you using?

By graphs do you mean my battery status page? If so, yes that is realtime. The HTML page is generated from data in the summary file by the php scripts batday.php and battery.php.

You could do the same with your graphs. IMO that is the best way as it means you can access the data from anywhere there is Internet access.

Simon

harmongt01, 2017-03-27:
I'm using LiveGraph.  An open source java application.  It should be able to parse certain date and timestamps but I don't think it is doing as well with this one.

I will take a look at what you've done.  On the tab named graphs I see you have a graph there that looks very nice.

George

simat, 2017-03-29:
Hi George,
From the look of the Github graphs I wouldn't be surprised if they are being generated by an open source program called gnuplot.

I currently have a weather station that is hooked up to the Internet which uses open source software called pywws that gets and logs the data from the weather station and generates graphs as JPEGs using gnuplot. Weatherbyyou [http://weatherbyyou.com/pywws/template/index.php?page=24h_graph](url) uses these JPEGs in webpages that are generated by PHP script files.

To generate graphs from my data I would probably use the program sed to strip out and edit the data I wanted to graph from the log file and then pass this on to gnuplot then use a php script to generate an HTML page.

Simon

harmongt01, 2017-03-29:
Yeah that is very cool Simon!  The site is very clean indeed!
I will look at these.  That may work better for me then.  
Had to work some long hours here the last few days so been out of the loop a bit.

George

simat, 2017-04-13:
Hi George,
I thought I would have a bit of a play around with gnuplot to plot data. For this graph I plotted the data from the daily log file. It is not very pretty but hopefully it gives you some ideas. Gnuplot is a very powerful and flexible plotting program!

[image]
To plot this I first wrote a shell script to strip out and reformat the data from the daily summary file generated by my battery monitor program. I have added this script file called strip.sh to the html directory.
To generate a file called plot ready for gnuplot type
./strip.sh [daysummary] [plot]
where daysummary is the path to your daysummary  file and plot is the name of the output file.

I then used gnuplot to generate the jpeg file. 
I wrote a script file called bargraph.plot which is also in the html directory to be parsed by gnuplot to generate the plot with the command
gnuplot bargraph.plot
The resulting file will be called plot.jpg. All these commands were carried out on the Beaglebone.

Have a look at the two script files and modify to suit your needs.

Hope this is of use.
Simon

harmongt01, 2017-04-29:
Fantastic Simon!!  I will certainly check this out.  We have some healthy storms blowing through today so it's interesting now to see the two wind turbines output with these wind gusts.  This afternoon I will try to recalibrate closer to the max current of the shunts as you suggested.

George

harmongt01, 2017-04-30:
Hey Simon,
Trying to run your strip.sh script but I think I'm doing something wrong:

root@beaglebone:~/Adafruit_Python_ADS1x15/BatteryMonitor/html# ls
bargraph.plot  batday.php  battery.php  SampleBatday.html  SampleBattery.html  strip.sh
root@beaglebone:~/Adafruit_Python_ADS1x15/BatteryMonitor/html# ./strip.sh /media/root/SD/daysummaryfile/daysummaryfile.txt plot.txt
bash: ./strip.sh: bash: bad interpreter: No such file or directory

harmongt01, 2017-05-01:
got it.

Changed the first line in the script from #! bash to  #!/bin/bash
I must have something set up differently.

closed 2017-12-05 by simat