simat/BatteryMonitor #9: New issue

harmongt01, 2017-12-05:
Hi Simon,

I had to reinstall everything.  I saved a backup but I think I'm missing something.  I get the following error when trying to import batteries:

root@beaglebone:/mnt/SD/BatteryMonitor# python
Python 2.7.13 (default, Jan 19 2017, 14:48:08)
[GCC 6.3.0 20170118] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import batteries
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "batteries.py", line 27, in <module>
    from config import loadconfig, config
ImportError: cannot import name loadconfig
>>>

Any ideas?

Thanks,
George

harmongt01, 2017-12-05:
[battery.cfg.txt](https://github.com/simat/BatteryMonitor/files/1529531/battery.cfg.txt)

simat, 2017-12-05:
Hi George,
Nice to hear you are still around.
I have done a major upgrade of the software which requires Python3.
Did you reclone my software form Github?
If so you will have to upgrade to Python 3.
You will also need to add the line  
**interface = 'from getatod import Raw'**  
to the [Files] section of battery.cfg

Simon

harmongt01, 2017-12-05:
Hi Simon,

I was not successful with upgrading to Python3 yet.  So I did cloned a fork I made.  This got me a little further now.  Is there any benefit for me to move to the newer code or resolve this as it is?  I think it gave me everything I needed.

root@beaglebone:/mnt/SD/BatteryMonitor# python
Python 2.7.13 (default, Jan 19 2017, 14:48:08)
[GCC 6.3.0 20170118] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import batteries
>>> batteries.deamon()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "batteries.py", line 124, in deamon
    logsummary.update(summary, batdata)
  File "summary.py", line 123, in update
    summary['current']['currentmax'][i] = round(batdata.currentav[i],1)
KeyError: 'currentmax'

Thanks,
George

simat, 2017-12-06:
Hi George,
That looks like an old version of the code before I modified it for your multiple power sources. 
What are the problems you are having with upgrading to Python3?

I found that Python3 was already preloaded on my Beaglebone. You just have to type the commend **python3** rather than **python**
I would get Python3 working and clone a fresh copy of my new code and use it with your **Battery.cfg** file with the extra line I stated in my last reply and the **summary.1cell**.

Simon

harmongt01, 2017-12-06:
Well you are right.  I remember running it in the past too now that you mentioned it.  So I will clone your latest code and add the line.  I'll let you know how that goes.

Thanks!
George

harmongt01, 2017-12-06:
Hi Simon,

Ok, one step forward, two steps back again :)

Python 3.5.3 (default, Jan 19 2017, 14:11:04)
[GCC 6.3.0 20170118] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import batteries
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/mnt/SD/BatteryMonitor/batteries.py", line 30, in <module>
    from getdata import Readings
  File "/mnt/SD/BatteryMonitor/getdata.py", line 22, in <module>
    exec(config['files']['interface'])
  File "<string>", line 1, in <module>
  File "/mnt/SD/BatteryMonitor/getatod.py", line 23, in <module>
    exec(i + '=' + config['AtoDs'][i])
  File "<string>", line 1, in <module>
  File "/mnt/SD/BatteryMonitor/ADS1x15.py", line 80, in __init__
    import Adafruit_GPIO.I2C as I2C
ImportError: No module named 'Adafruit_GPIO'
>>>

I reinstalled everything from scratch so I may not have this module installed in the right place.

harmongt01, 2017-12-06:
I found where someone had a similar problem.   They reinstalled Adafruit_Python_GPIO using python3 so I tried that.  Now the error is different but maybe I should hold here before I get too deep in trouble:

Python 3.5.3 (default, Jan 19 2017, 14:11:04)
[GCC 6.3.0 20170118] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import batteries
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/mnt/SD/BatteryMonitor/batteries.py", line 30, in <module>
    from getdata import Readings
  File "/mnt/SD/BatteryMonitor/getdata.py", line 22, in <module>
    exec(config['files']['interface'])
  File "<string>", line 1, in <module>
  File "/mnt/SD/BatteryMonitor/getatod.py", line 23, in <module>
    exec(i + '=' + config['AtoDs'][i])
  File "<string>", line 1, in <module>
  File "/mnt/SD/BatteryMonitor/ADS1x15.py", line 82, in __init__
    self._device = i2c.get_i2c_device(address, **kwargs)
  File "/usr/local/lib/python3.5/dist-packages/Adafruit_GPIO-1.0.3-py3.5.egg/Adafruit_GPIO/I2C.py", line 65, in get_i2c_device
TypeError: __init__() got an unexpected keyword argument 'ic'

simat, 2017-12-06:
I had forgotten about having to reload the Adafruit library, I will update my documentation.

[Here is the link](https://learn.adafruit.com/raspberry-pi-analog-to-digital-converters/ads1015-slash-ads1115) with instructions on how to install the Adafruit python3 library.

harmongt01, 2017-12-07:
Hi Simon,

I tried the steps but I am having some issue.  Keep running into some dependency problem with chromium that seems to be causing a problem:

root@beaglebone:~# git clone https://github.com/adafruit/Adafruit_Python_ADS1x15                                                                      .git
Cloning into 'Adafruit_Python_ADS1x15'...
remote: Counting objects: 52, done.
remote: Total 52 (delta 0), reused 0 (delta 0), pack-reused 52
Unpacking objects: 100% (52/52), done.
root@beaglebone:~# cd Adafruit_Python_ADS1x15/
root@beaglebone:~/Adafruit_Python_ADS1x15# python setup.py install
running install
running bdist_egg
running egg_info
creating Adafruit_ADS1x15.egg-info
writing requirements to Adafruit_ADS1x15.egg-info/requires.txt
writing Adafruit_ADS1x15.egg-info/PKG-INFO
writing top-level names to Adafruit_ADS1x15.egg-info/top_level.txt
writing dependency_links to Adafruit_ADS1x15.egg-info/dependency_links.txt
writing manifest file 'Adafruit_ADS1x15.egg-info/SOURCES.txt'
reading manifest file 'Adafruit_ADS1x15.egg-info/SOURCES.txt'
writing manifest file 'Adafruit_ADS1x15.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-armv7l/egg
running install_lib
running build_py
creating build
creating build/lib.linux-armv7l-2.7
creating build/lib.linux-armv7l-2.7/Adafruit_ADS1x15
copying Adafruit_ADS1x15/__init__.py -> build/lib.linux-armv7l-2.7/Adafruit_ADS1                                                                      x15
copying Adafruit_ADS1x15/ADS1x15.py -> build/lib.linux-armv7l-2.7/Adafruit_ADS1x                                                                      15
creating build/bdist.linux-armv7l
creating build/bdist.linux-armv7l/egg
creating build/bdist.linux-armv7l/egg/Adafruit_ADS1x15
copying build/lib.linux-armv7l-2.7/Adafruit_ADS1x15/__init__.py -> build/bdist.l                                                                      inux-armv7l/egg/Adafruit_ADS1x15
copying build/lib.linux-armv7l-2.7/Adafruit_ADS1x15/ADS1x15.py -> build/bdist.li                                                                      nux-armv7l/egg/Adafruit_ADS1x15
byte-compiling build/bdist.linux-armv7l/egg/Adafruit_ADS1x15/__init__.py to __in                                                                      it__.pyc
byte-compiling build/bdist.linux-armv7l/egg/Adafruit_ADS1x15/ADS1x15.py to ADS1x                                                                      15.pyc
creating build/bdist.linux-armv7l/egg/EGG-INFO
copying Adafruit_ADS1x15.egg-info/PKG-INFO -> build/bdist.linux-armv7l/egg/EGG-I                                                                      NFO
copying Adafruit_ADS1x15.egg-info/SOURCES.txt -> build/bdist.linux-armv7l/egg/EG                                                                      G-INFO
copying Adafruit_ADS1x15.egg-info/dependency_links.txt -> build/bdist.linux-armv                                                                      7l/egg/EGG-INFO
copying Adafruit_ADS1x15.egg-info/requires.txt -> build/bdist.linux-armv7l/egg/E                                                                      GG-INFO
copying Adafruit_ADS1x15.egg-info/top_level.txt -> build/bdist.linux-armv7l/egg/                                                                      EGG-INFO
zip_safe flag not set; analyzing archive contents...
creating dist
creating 'dist/Adafruit_ADS1x15-1.0.2-py2.7.egg' and adding 'build/bdist.linux-a                                                                      rmv7l/egg' to it
removing 'build/bdist.linux-armv7l/egg' (and everything under it)
Processing Adafruit_ADS1x15-1.0.2-py2.7.egg
Removing /usr/local/lib/python2.7/dist-packages/Adafruit_ADS1x15-1.0.2-py2.7.egg
Copying Adafruit_ADS1x15-1.0.2-py2.7.egg to /usr/local/lib/python2.7/dist-packag                                                                      es
Adafruit-ADS1x15 1.0.2 is already the active version in easy-install.pth

Installed /usr/local/lib/python2.7/dist-packages/Adafruit_ADS1x15-1.0.2-py2.7.eg                                                                      g
Processing dependencies for Adafruit-ADS1x15==1.0.2
Searching for Adafruit-GPIO==1.0.3
Best match: Adafruit-GPIO 1.0.3
Processing Adafruit_GPIO-1.0.3-py2.7.egg
Adafruit-GPIO 1.0.3 is already the active version in easy-install.pth

Using /usr/local/lib/python2.7/dist-packages/Adafruit_GPIO-1.0.3-py2.7.egg
Searching for spidev==3.2
Best match: spidev 3.2
Processing spidev-3.2-py2.7-linux-armv7l.egg
spidev 3.2 is already the active version in easy-install.pth

Using /usr/local/lib/python2.7/dist-packages/spidev-3.2-py2.7-linux-armv7l.egg
Searching for Adafruit-PureIO==0.2.1
Best match: Adafruit-PureIO 0.2.1
Processing Adafruit_PureIO-0.2.1-py2.7.egg
Adafruit-PureIO 0.2.1 is already the active version in easy-install.pth
[... 18 lines cut ...]
Requirement already satisfied: Adafruit-GPIO>=0.6.5 in /usr/local/lib/python2.7/                                                                      dist-packages/Adafruit_GPIO-1.0.3-py2.7.egg (from adafruit-ads1x15)
Requirement already satisfied: adafruit-pureio in /usr/local/lib/python2.7/dist-                                                                      packages/Adafruit_PureIO-0.2.1-py2.7.egg (from Adafruit-GPIO>=0.6.5->adafruit-ad                                                                      s1x15)
[... cut ...]
  File "/mnt/SD/BatteryMonitor/getatod.py", line 23, in <module>
    exec(i + '=' + config['AtoDs'][i])
  File "<string>", line 1, in <module>
  File "/mnt/SD/BatteryMonitor/ADS1x15.py", line 82, in __init__
    self._device = i2c.get_i2c_device(address, **kwargs)
  File "/usr/local/lib/python3.5/dist-packages/Adafruit_GPIO-1.0.3-py3.5.egg/Ada                                                                      fruit_GPIO/I2C.py", line 65, in get_i2c_device
TypeError: __init__() got an unexpected keyword argument 'ic'
>>>

simat, 2017-12-07:
Try changing 
**root@beaglebone:~/Adafruit_Python_ADS1x15# python setup.py install**
to
**root@beaglebone:~/Adafruit_Python_ADS1x15# python3 setup.py **install**
If this fails try the **Python Package Index Install** instructions further down the Adafruit installation page instructions.

harmongt01, 2017-12-07:
Tried the command you suggested and got the following result when trying to run.

root@beaglebone:~/Adafruit_Python_ADS1x15# cd /mnt/SD/BatteryMonitor/
root@beaglebone:/mnt/SD/BatteryMonitor# python3
Python 3.5.3 (default, Jan 19 2017, 14:11:04)
[GCC 6.3.0 20170118] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import batteries
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/mnt/SD/BatteryMonitor/batteries.py", line 30, in <module>
    from getdata import Readings
  File "/mnt/SD/BatteryMonitor/getdata.py", line 22, in <module>
    exec(config['files']['interface'])
  File "<string>", line 1, in <module>
  File "/mnt/SD/BatteryMonitor/getatod.py", line 23, in <module>
    exec(i + '=' + config['AtoDs'][i])
  File "<string>", line 1, in <module>
  File "/mnt/SD/BatteryMonitor/ADS1x15.py", line 82, in __init__
    self._device = i2c.get_i2c_device(address, **kwargs)
  File "/usr/local/lib/python3.5/dist-packages/Adafruit_GPIO-1.0.3-py3.5.egg/Adafruit_GPIO/I2C.py", line 65, in get_i2c_device
TypeError: __init__() got an unexpected keyword argument 'debug'

I will Python Package Index Install instructions tomorrow.

Thanks,
George

simat, 2017-12-07:
Ah, I remember now, Adafruit have changed the driver.

To work with the new driver try your battery.cfg that I have modified. 
[batteryGeorge.cfg.txt](https://github.com/simat/BatteryMonitor/files/1537756/batteryGeorge.cfg.txt)

harmongt01, 2017-12-07:
results are different:

root@beaglebone:/mnt/SD/BatteryMonitor# python3
Python 3.5.3 (default, Jan 19 2017, 14:11:04)
[GCC 6.3.0 20170118] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import batteries
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/mnt/SD/BatteryMonitor/batteries.py", line 30, in <module>
    from getdata import Readings
  File "/mnt/SD/BatteryMonitor/getdata.py", line 22, in <module>
    exec(config['files']['interface'])
KeyError: 'interface'

simat, 2017-12-07:
Sorry, I left out the Interface line in the Files section.

You will need to add the line
**interface = 'from getatod import Raw'**
to the [Files] section of battery.cfg

simat, 2017-12-07:
I have finally had a look at the how to do proper versioning with Git.
You will see there are now **3 releases** on a link on the **code** page. Each one of these is a snapshot at a particular time.

You can always go back to any point using the **commits** link

harmongt01, 2017-12-08:
I added the interface line to my saved backup file so I don't keep running into that.

Do you think I should try one of the releases you added?  After adding the interface line to battery.cfg I am back to getting the following error:

Python 3.5.3 (default, Jan 19 2017, 14:11:04)
[GCC 6.3.0 20170118] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import batteries
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/mnt/SD/BatteryMonitor/batteries.py", line 30, in <module>
    from getdata import Readings
  File "/mnt/SD/BatteryMonitor/getdata.py", line 22, in <module>
    exec(config['files']['interface'])
  File "<string>", line 1, in <module>
  File "/mnt/SD/BatteryMonitor/getatod.py", line 23, in <module>
    exec(i + '=' + config['AtoDs'][i])
  File "<string>", line 1, in <module>
  File "/mnt/SD/BatteryMonitor/ADS1x15.py", line 82, in __init__
    self._device = i2c.get_i2c_device(address, **kwargs)
  File "/usr/local/lib/python3.5/dist-packages/Adafruit_GPIO-1.0.3-py3.5.egg/Adafruit_GPIO/I2C.py", line 65, in get_i2c_device
TypeError: __init__() got an unexpected keyword argument 'ic'

simat, 2017-12-08:
Hi George,
Hopefully we are both online.
You have to use the batteryGeorge.cfg.txt that I attached to a day ago.

Add the interface line to that file and rename it battery.cfg

harmongt01, 2017-12-08:
Hi Simon,

I am using the batteryGeorge.cfg file you sent me.  But I get that same error.

Python 3.5.3 (default, Jan 19 2017, 14:11:04)
[GCC 6.3.0 20170118] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import batteries
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/mnt/SD/BatteryMonitor/batteries.py", line 30, in <module>
    from getdata import Readings
  File "/mnt/SD/BatteryMonitor/getdata.py", line 22, in <module>
    exec(config['files']['interface'])
  File "<string>", line 1, in <module>
  File "/mnt/SD/BatteryMonitor/getatod.py", line 23, in <module>
    exec(i + '=' + config['AtoDs'][i])
  File "<string>", line 1, in <module>
  File "/mnt/SD/BatteryMonitor/ADS1x15.py", line 82, in __init__
    self._device = i2c.get_i2c_device(address, **kwargs)
  File "/usr/local/lib/python3.5/dist-packages/Adafruit_GPIO-1.0.3-py3.5.egg/Adafruit_GPIO/I2C.py", line 65, in get_i2c_device
TypeError: __init__() got an unexpected keyword argument 'ic'

simat, 2017-12-08:
I edited my last comment and added 
"I just had a look at the config file again. Sorry I have made another mistake, need a new brain!
change this line
atod1 = 'AtoD.ADS1x15(address=0x49,ic=0x01,busnum=2)'
to
atod1 = 'AtoD.ADS1x15(address=0x49,busnum=2)'
"

That should fix it, as you can see from the last line of the error it is getting an unexpected arguement 'ic'

simat, 2017-12-09:
Hi George,
It is a little hard for me to test your configuration but I have loaded my BMS software with your config file.

I have found some other errors. The attached file hopefully has fixed all the problems [batteryGeorge.cfg.txt](https://github.com/simat/BatteryMonitor/files/1544274/batteryGeorge.cfg.txt)
It now compiles on my computer but gives an error when it tries to access the non existent I2C bus on my computer.

So hopefully this is it!
Simon

harmongt01, 2017-12-09:
HI Simon,

By the way thanks again for helping me out with this.

A couple steps forward I think.  Now it gives the following error about GPIO

Python 3.5.3 (default, Jan 19 2017, 14:11:04)
[GCC 6.3.0 20170118] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import batteries
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/mnt/SD/BatteryMonitor/batteries.py", line 34, in <module>
    exec(config['alarms'][i][0])
  File "<string>", line 1, in <module>
NameError: name 'GPIO' is not defined

simat, 2017-12-10:
Hi George,
Adafruit have changed the way they do IO on the Beaglebone and I am having a battle to get it working.

There is currently a test alarm
**alarm = ['GPIO.setup("P8_10", GPIO.OUT)', 'test=minvolts<2.8 or maxvolts>3.60', 'GPIO.output("P8_10", GPIO.HIGH)', \ 
	'test=minvolts>2.9 and maxvolts<3.59', 'GPIO.output("P8_10", GPIO.LOW)']**

In the alarms section of the config file. I assume you are not using this, if so you could delete it or comment it out. This should solve the GPIO problem.

harmongt01, 2017-12-10:
I am having this strange chromium dependency problem that seems to be causing problems to install other things like jre.

root@beaglebone:/home/gharmon# apt-get install default-jre
Reading package lists... Done
Building dependency tree
Reading state information... Done
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 chromium : Depends: chromium-common (= 60.0.3112.78-1rcnee0~stretch+20170806) but 61.0.3163.100-2rcnee0~stretch+20170927 is to be installed
 chromium-l10n : Depends: chromium (>= 62.0.3202.89-1~deb9u1) but 60.0.3112.78-1rcnee0~stretch+20170806 is to be installed
 default-jre : Depends: default-jre-headless (= 2:1.8-58)
               Depends: openjdk-8-jre
               Recommends: default-java-plugin but it is not going to be installed
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
root@beaglebone:/home/gharmon#

harmongt01, 2017-12-10:
Hi Simon,

This is after commenting out the alarm you mentioned.

root@beaglebone:/mnt/SD/BatteryMonitor# python3
Python 3.5.3 (default, Jan 19 2017, 14:11:04)
[GCC 6.3.0 20170118] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import batteries
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/mnt/SD/BatteryMonitor/batteries.py", line 27, in <module>
    from config import loadconfig, config
  File "/mnt/SD/BatteryMonitor/config.py", line 28, in <module>
    loadconfig()
  File "/mnt/SD/BatteryMonitor/config.py", line 26, in loadconfig
    config[section][key] = literal_eval(val)
  File "/usr/lib/python3.5/ast.py", line 46, in literal_eval
    node_or_string = parse(node_or_string, mode='eval')
  File "/usr/lib/python3.5/ast.py", line 35, in parse
    return compile(source, filename, mode, PyCF_ONLY_AST)
  File "<unknown>", line 1
    minvolts>2.9 and maxvolts<3.59', 'GPIO.output("P8_10", GPIO.LOW)']
                                     ^
SyntaxError: invalid syntax
>>>

I must have multiple issues :)

harmongt01, 2017-12-10:
Hi Simon,

Now that I have your program running, I see the voltage reading is now in the 9V range but it should be in the 13v range.  I notice the changes to your cfg file but not sure I understand how to properly calibrate things.

For example the old AtoDs and VoltageInputs were the following:
[AtoDs]
atod0 = 'AtoD.ADS1x15(busnum=2,address=0x48,ic=0x01, debug=True)'
atod1 = 'AtoD.ADS1x15(busnum=2,address=0x49,ic=0x01, debug=True)'
atod2 = 'AtoD.ADS1x15(busnum=2,address=0x4A,ic=0x01, debug=True)'

[VoltageInputs]
v1 = 'atod0.readADCSingleEnded(channel=0, pga=4096, sps=250)'

The new ones are:
[AtoDs]
atod0 = 'AtoD.ADS1115(address=0x48,busnum=2)'
atod1 = 'AtoD.ADS1115(address=0x49,busnum=2)'
atod2 = 'AtoD.ADS1115(address=0x4A,busnum=2)'

[VoltageInputs]
v1 = 'atod0.read_adc(0, gain=2)*0.06250197'

These changes are the same in CurrentInputs also.  I read through some of your updates to the instructions area of github but I'm not quite up to speed on it.  Maybe it's still a work in progress.

What steps should I follow to adjust voltage?  I will want to calibrate the current as well after that.  Will that calibration still be similar as before?

Thanks,
George

simat, 2017-12-11:
Hi George,

I have hopefully made the calibration easier with three new programs. See the Calibrate wiki page to get details.

Simon

harmongt01, 2017-12-12:
Hey Simon,

Something is not right with my voltage measurement.  My wiring and voltage divider should be exactly the same as before.  Now the voltage is being recorded by the program as always 9.391 with no fluctuation as if it is not reading anything from the batteries.  When I disconnect the positive lead going from my voltage divider to the ADS then the voltage reading changes to another constant of 0. something.  So the lead from the voltage divider is providing something.  

Also, I do not really understand how to use your calibration to adjust the voltage measurement.

But it looks like I have two issues.  First, the voltage should always be fluctuating even if the reading is not the expected 13.x volts.  Second, how do I use the calibration utility?

From your calibration instructions:
3. Firstly perform the course voltage with the command python3 calvcourse.py
Measure with an accurate multimeter the value of each battery cell relative to the battery negative. To improve the overall accuracy make sure that all the readings taken with the multimeter are done on the same range.

How do I input the actual reading from the voltmeter?  When I send the command 'python3 calvcourse.py' it returns me to the cursor without any prompt to enter a value.

Thanks,
George

simat, 2017-12-14:
I have started a  new issue called #10

closed 2017-12-14 by simat