simat/BatteryMonitor #10: Calibration

simat, 2017-12-14:
> Hey Simon,
> 
> Something is not right with my voltage measurement. My wiring and voltage divider should be exactly the same as before. Now the voltage is being recorded by the program as always 9.391 with no fluctuation as if it is not reading anything from the batteries. When I disconnect the positive lead going from my voltage divider to the ADS then the voltage reading changes to another constant of 0. something. So the lead from the voltage divider is providing something.
> 
> Also, I do not really understand how to use your calibration to adjust the voltage measurement.
> 
> But it looks like I have two issues. First, the voltage should always be fluctuating even if the reading is not the expected 13.x volts. Second, how do I use the calibration utility?
> 
> From your calibration instructions:
> 3. Firstly perform the course voltage with the command python3 calvcourse.py
> Measure with an accurate multimeter the value of each battery cell relative to the battery negative. To improve the overall accuracy make sure that all the readings taken with the multimeter are done on the same range.
> 
> How do I input the actual reading from the voltmeter? When I send the command 'python3 calvcourse.py' it returns me to the cursor without any prompt to enter a value.
> 
> Thanks,
> George

Hi George,

The calibration software uses the last ten entries from the Battery Monitor output log so the first thing to check would be the output is being generated correctly.

You can check this with the shell command
**tail /mnt/SD/logfile/logfile.txt**

Before doing this do a git pull as I have fixed a bug to make sure that the log file is written to disk after each update.

Simon

simat, 2017-12-14:
If you use 
**tail /mnt/SD/logfile/logfile.txt -f**
You will see the logfile being updated every ten seconds

harmongt01, 2017-12-19:
I did the git pull and there were a handful of files updated.  I must be off on some update or two with software and config.  Should be getting close.
Now when I run I get the following error:

root@beaglebone:/mnt/SD/BatteryMonitor# python3
Python 3.5.3 (default, Jan 19 2017, 14:11:04)
[GCC 6.3.0 20170118] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import batteries
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/mnt/SD/BatteryMonitor/batteries.py", line 30, in <module>
    from getdata import Readings
  File "/mnt/SD/BatteryMonitor/getdata.py", line 22, in <module>
    exec(config['files']['interface'])
  File "<string>", line 1, in <module>
  File "/mnt/SD/BatteryMonitor/getatod.py", line 23, in <module>
    exec(i + '=' + config['AtoDs'][i])
  File "<string>", line 1, in <module>
NameError: name 'AtoD' is not defined

simat, 2017-12-20:
I have moved your last comment to a new issue. I can't work out how to just move the comment, have to copy and paste

Simon

closed 2020-04-11 by simat