simat/BatteryMonitor #12: Upgrading to new version

simat, 2017-12-20:
> I did the git pull and there were a handful of files updated.  I must be off on some update or two with software and config.  Should be getting close.
> Now when I run I get the following error:
> 
> root@beaglebone:/mnt/SD/BatteryMonitor# python3
> Python 3.5.3 (default, Jan 19 2017, 14:11:04)
> [GCC 6.3.0 20170118] on linux
> Type "help", "copyright", "credits" or "license" for more information.
> >>> import batteries
> Traceback (most recent call last):
>   File "<stdin>", line 1, in <module>
>   File "/mnt/SD/BatteryMonitor/batteries.py", line 30, in <module>
>     from getdata import Readings
>   File "/mnt/SD/BatteryMonitor/getdata.py", line 22, in <module>
>     exec(config['files']['interface'])
>   File "<string>", line 1, in <module>
>   File "/mnt/SD/BatteryMonitor/getatod.py", line 23, in <module>
>     exec(i + '=' + config['AtoDs'][i])
>   File "<string>", line 1, in <module>
> NameError: name 'AtoD' is not defined

Hi George,

It looks like you are missing the **AtoDs** section in the battery.cfg file. 

Whenever you clone the software, make a new copy  or upgrade the software you should make sure you copy your existing **battery.cfg** file to the same directory as the Battery Monitor software which is setup for your hardware configuration. You might have to make changes to this file to work with the newer version of the software. You can use the different variants of the sample battery configuration file that I supply as a guide to see what needs to be changed in your config file. 

This file 
[batteryGeorge.cfg.txt](https://github.com/simat/BatteryMonitor/files/1575211/batteryGeorge.cfg.txt)
should I hope be the one that will work with your hardware.

Simon

harmongt01, 2017-12-29:
Hi Simon,

Happy Holidays to you and your family!

I updated the battery.cfg.  It is in the BatteryMonitor directory.  
The program runs now.  I am still seeing the same issue where the voltage reading remains a constant value (9.679) with no fluctuation.  The current measurements may be working ok.  The program stops at some point with what looks to be a timeout.  I assume it must be related to the voltage reading.  It seems the ADS board is not returning the actual voltage reading and your gain is applied which returns this constant value I'm seeing.  Not quite sure what the problem is yet.

9.679 9.730 4.679 0.900 0.000 0.100 0.000 41.410 46.150
9.679 9.730 4.679 0.900 0.000 0.100 0.000 41.410 46.150
9.679 9.730 4.679 0.900 0.000 0.100 0.000 41.420 46.150
9.679 9.730 4.679 0.900 0.000 0.100 0.000 41.420 46.150
9.679 9.730 4.679 0.900 0.000 0.100 0.000 41.420 46.160
9.679 9.730 4.679 0.900 0.000 0.100 0.000 41.420 46.160
9.679 9.730 4.679 0.900 0.000 0.100 0.000 41.430 46.160
9.679 9.730 4.679 0.900 0.000 0.100 0.000 41.430 46.160
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/mnt/SD/BatteryMonitor/batteries.py", line 80, in deamon
    batdata.getraw()
  File "/mnt/SD/BatteryMonitor/getdata.py", line 101, in getraw
    self.getvi()
  File "/mnt/SD/BatteryMonitor/getdata.py", line 78, in getvi
    self.rawdata.x()
  File "/mnt/SD/BatteryMonitor/getatod.py", line 44, in x
    self.rawv[i+1] = eval(self.vin[i])/1000 # A to D 1 to 4 in volts
  File "<string>", line 1, in <module>
  File "/usr/local/lib/python3.5/dist-packages/Adafruit_ADS1x15-1.0.2-py3.5.egg/Adafruit_ADS1x15/ADS1x15.py", line 192, in read_adc
  File "/usr/local/lib/python3.5/dist-packages/Adafruit_ADS1x15-1.0.2-py3.5.egg/Adafruit_ADS1x15/ADS1x15.py", line 128, in _read
  File "/usr/local/lib/python3.5/dist-packages/Adafruit_GPIO-1.0.3-py3.5.egg/Adafruit_GPIO/I2C.py", line 128, in writeList
  File "/usr/local/lib/python3.5/dist-packages/Adafruit_PureIO-0.2.1-py3.5.egg/Adafruit_PureIO/smbus.py", line 274, in write_i2c_block_data
TimeoutError: [Errno 110] Connection timed out
>>>

It looks to be a problem maybe only with the voltage reading.  The voltage is present.

simat, 2018-01-01:
Hi George,
Happy New Year!
I think the Timeout error is a problem with the Linux I2C driver, hopefully it is a one off. I don't think it is anything to do with the voltage problem.

There is a test program called simpletest.py that is part of the ADS1x15 package which I have pasted below. Run this and see what results you get.

```
# Simple demo of reading each analog input from the ADS1x15 and printing it to
# the screen.
# Author: Tony DiCola
# License: Public Domain
import time

# Import the ADS1x15 module.
import Adafruit_ADS1x15

# Create an ADS1115 ADC (16-bit) instance.
adc = Adafruit_ADS1x15.ADS1115()

# Or create an ADS1015 ADC (12-bit) instance.
#adc = Adafruit_ADS1x15.ADS1015()

# Note you can change the I2C address from its default (0x48), and/or the I2C
# bus by passing in these optional parameters:
#adc = Adafruit_ADS1x15.ADS1015(address=0x49, busnum=1)

# Choose a gain of 1 for reading voltages from 0 to 4.09V.
# Or pick a different gain to change the range of voltages that are read:
#  - 2/3 = +/-6.144V
#  -   1 = +/-4.096V
#  -   2 = +/-2.048V
#  -   4 = +/-1.024V
#  -   8 = +/-0.512V
#  -  16 = +/-0.256V
# See table 3 in the ADS1015/ADS1115 datasheet for more info on gain.
GAIN = 1

print('Reading ADS1x15 values, press Ctrl-C to quit...')
# Print nice channel column headers.
print('| {0:>6} | {1:>6} | {2:>6} | {3:>6} |'.format(*range(4)))
print('-' * 37)
# Main loop.
while True:
    # Read all the ADC channel values in a list.
    values = [0]*4
    for i in range(4):
        # Read the specified ADC channel using the previously set gain value.
        values[i] = adc.read_adc(i, gain=GAIN)
        # Note you can also pass in an optional data_rate parameter that controls
        # the ADC conversion time (in samples/second). Each chip has a different
        # set of allowed data rate values, see datasheet Table 9 config register
        # DR bit values.
        #values[i] = adc.read_adc(i, gain=GAIN, data_rate=128)
        # Each value will be a 12 or 16 bit signed integer value depending on the
        # ADC (ADS1015 = 12-bit, ADS1115 = 16-bit).
    # Print the ADC values.
    print('| {0:>6} | {1:>6} | {2:>6} | {3:>6} |'.format(*values))
    # Pause for half a second.
    time.sleep(0.5)
```

harmongt01, 2018-01-02:
Hi Simon,

This is what I received:

root@beaglebone:/mnt/SD/BatteryMonitor# python3
Python 3.5.3 (default, Jan 19 2017, 14:11:04)
[GCC 6.3.0 20170118] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import time
>>> import Adafruit_ADS1x15
>>> adc = Adafruit_ADS1x15.ADS1115()
>>> GAIN = 1
>>> print('Reading ADS1x15 values, press Ctrl-C to quit...')
Reading ADS1x15 values, press Ctrl-C to quit...
>>> print('| {0:>6} | {1:>6} | {2:>6} | {3:>6} |'.format(*range(4)))
|      0 |      1 |      2 |      3 |
>>> print('-' * 37)
-------------------------------------
>>> while True:
... # Read all the ADC channel values in a list.
...     values = [0]*4
...     for i in range(4):
        values[i] = adc.read_adc(i, gain=GAIN)
...         # Read the specified ADC channel using the previously set gain value.
...         values[i] = adc.read_adc(i, gain=GAIN)
...         # Note you can also pass in an optional data_rate parameter that controls
...         # the ADC conversion time (in samples/second). Each chip has a different
...         # set of allowed data rate values, see datasheet Table 9 config register
...         # DR bit values.
...         #values[i] = adc.read_adc(i, gain=GAIN, data_rate=128)
...         # Each value will be a 12 or 16 bit signed integer value depending on the
...         # ADC (ADS1015 = 12-bit, ADS1115 = 16-bit).
...     # Print the ADC values.
...     print('| {0:>6} | {1:>6} | {2:>6} | {3:>6} |'.format(*values))
...     # Pause for half a second.
...     time.sleep(0.5)
...
Traceback (most recent call last):
  File "<stdin>", line 6, in <module>
  File "/usr/local/lib/python3.5/dist-packages/Adafruit_ADS1x15-1.0.2-py3.5.egg/Adafruit_ADS1x15/ADS1x15.py", line 192, in read_adc
  File "/usr/local/lib/python3.5/dist-packages/Adafruit_ADS1x15-1.0.2-py3.5.egg/Adafruit_ADS1x15/ADS1x15.py", line 128, in _read
  File "/usr/local/lib/python3.5/dist-packages/Adafruit_GPIO-1.0.3-py3.5.egg/Adafruit_GPIO/I2C.py", line 128, in writeList
  File "/usr/local/lib/python3.5/dist-packages/Adafruit_PureIO-0.2.1-py3.5.egg/Adafruit_PureIO/smbus.py", line 274, in write_i2c_block_data
TimeoutError: [Errno 110] Connection timed out
>>>

harmongt01, 2018-01-02:
Simon,

I am not sure if this is the problem, but I have been having trouble to update certain things because of some missing dependencies.  See what happens when I try to do an apt-get install:

root@beaglebone:/mnt/SD# sudo apt-get install git build-essential python-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
build-essential is already the newest version (12.3).
git is already the newest version (1:2.11.0-3+deb9u2).
python-dev is already the newest version (2.7.13-2).
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 chromium : Depends: chromium-common (= 60.0.3112.78-1rcnee0~stretch+20170806) but 61.0.3163.100-2rcnee0~stretch+20170927 is to be installed
 chromium-l10n : Depends: chromium (>= 62.0.3202.89-1~deb9u1) but 60.0.3112.78-1rcnee0~stretch+20170806 is to be installed
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
root@beaglebone:/mnt/SD#

When I try to run the apt --fix-broken install the following results:

root@beaglebone:/mnt/SD# apt --fix-broken install
Reading package lists... Done
Building dependency tree
Reading state information... Done
Correcting dependencies... Done
The following package was automatically installed and is no longer required:
  chromium-common
Use 'apt autoremove' to remove it.
The following additional packages will be installed:
  chromium
Suggested packages:
  chromium-shell chromium-driver chromium-widevine
The following packages will be upgraded:
  chromium
1 upgraded, 0 newly installed, 0 to remove and 27 not upgraded.
2 not fully installed or removed.
Need to get 44.5 MB of archives.
After this operation, 16.8 MB of additional disk space will be used.
Do you want to continue? [Y/n] Y
Get:1 http://deb.debian.org/debian stretch/main armhf chromium armhf 62.0.3202.89-1~deb9u1 [44.5 MB]
Fetched 44.5 MB in 38s (1,149 kB/s)
(Reading database ... 76269 files and directories currently installed.)
Preparing to unpack .../chromium_62.0.3202.89-1~deb9u1_armhf.deb ...
Unpacking chromium (62.0.3202.89-1~deb9u1) over (60.0.3112.78-1rcnee0~stretch+20170806) ...
dpkg: error processing archive /var/cache/apt/archives/chromium_62.0.3202.89-1~deb9u1_armhf.deb (--unpack):
 trying to overwrite '/usr/lib/chromium/natives_blob.bin', which is also in package chromium-common 61.0.3163.100-2rcnee0~stretch+20170927
dpkg-deb: error: subprocess paste was killed by signal (Broken pipe)
Errors were encountered while processing:
 /var/cache/apt/archives/chromium_62.0.3202.89-1~deb9u1_armhf.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
root@beaglebone:/mnt/SD#

I ran into this issue at some point after rebuilding everything from scratch but I am not sure yet how to fix it.  My feeling is it could be the reason some things are not installed/updated correctly.

harmongt01, 2018-01-03:
Again, no rush.  I appreciate all your help and experience.  We are in winter here so my aquaponic greenhouse is on hold till spring.  I plan to eventually tie in the alternative energy to that and not use grid power.

Thanks,
George

simat, 2018-01-04:
Hi George,
I am happy to help, I just hope it is not too frustrating for you. Sometimes these dependency issues can be a problem

I am fairly sure that all the stuff about chromium is about the chromium web browser so can be ignored.

The timeout error is another matter, is the timeout error occurring with my software as well?

Try running the adafruit software with version 2 Python by typing python not python3

Simon

harmongt01, 2018-01-09:
Hi Simon,

I still get a timeout with version 2 python:

Traceback (most recent call last):
  File "<stdin>", line 7, in <module>
  File "build/bdist.linux-armv7l/egg/Adafruit_ADS1x15/ADS1x15.py", line 192, in                                                                                                                                read_adc
  File "build/bdist.linux-armv7l/egg/Adafruit_ADS1x15/ADS1x15.py", line 128, in                                                                                                                                _read
  File "build/bdist.linux-armv7l/egg/Adafruit_GPIO/I2C.py", line 128, in writeLi                                                                                                                               st
  File "build/bdist.linux-armv7l/egg/Adafruit_PureIO/smbus.py", line 274, in wri                                                                                                                               te_i2c_block_data
IOError: [Errno 110] Connection timed out

Maybe I don't have the Adafruit ADS software installed properly.  Does it matter what directory I have it installed under?  I installed the ADS software under /root and the BatteryMonitor I have installed under another directory.  Probably doesn't matter but I thought I'd ask.  My Unix/Linux experience is pretty basic.  Seems like a driver issue like improper installation perhaps.

Thanks,
George

harmongt01, 2018-01-11:
If I run the i2cdetect I I see it detects no interfaces:

root@beaglebone:/home/gharmon# i2cdetect -r 1
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will probe file /dev/i2c-1 using read byte commands.
I will probe address range 0x03-0x77.
Continue? [Y/n] Y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

harmongt01, 2018-01-11:
Wait... I spoke too soon:

root@beaglebone:/home/gharmon# i2cdetect -r 2
WARNING! This program can confuse your I2C bus, cause data loss and worse!
I will probe file /dev/i2c-2 using read byte commands.
I will probe address range 0x03-0x77.
Continue? [Y/n] Y
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- 48 49 4a -- -- -- -- --
50: -- -- -- -- UU UU UU UU -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

simat, 2018-01-12:
I would try to reinstall The Adafruit drivers from [https://github.com/adafruit/Adafruit_Python_ADS1x15](url) and follow the installation instructions.
For python3 during the install type
'sudo python3 setup.py install`
rather than sudo python setup.py install

You might be right that it is something to do with permissions although I haven't had any problems with this. You could try running python as root with sudo.

Simon

bres55, 2018-01-12:
Not really an issue, but a question. Did you ever try to just read the bms board with modbus?
I have one of these
https://www.aliexpress.com/item/8S-60A-Li-ion-LiPo-BMS-PCM-PCB-battery-protection-board-for-8-Packs-18650-Battery/32844170262.html,
connected to one of these
https://www.ebay.co.uk/itm/6Pin-USB-2-0-to-TTL-UART-Module-Converter-CP2102-STC-Replace-FT232-CF/272249732398?epid=503069058&hash=item3f63593d2e:g:QVUAAOSw71BXP92B
and use this
http://www.wlcsystems.com/download.html
to read it, holding register 03,
reg addresses, 0 to 20 gives me COVP down to DSGOC,
I have an 8S 40Amp bms, and so registers 21 to 28 give me the cell voltages.
I am hoping to hook this to my arduino, as I already read, using modbus, a 
https://www.aliexpress.com/item/Single-Phase-Multi-Function-For-V-A-Hz-PF-Dmd-KWh-KVarh-Din-Rail-Digital-Energy/32457007359.html
and a Tracer3215BN with modbus.
I have tried reading the bms using other modbus function registers but no joy. Hence the reason I am asking if you explored this avenue. 
Thanks for putting your work on github.

harmongt01, 2018-01-12:
Hi bre55,

Before I started using Simons BatteryMonitor solution I only had digital guages which do not allow me to record historical data.  For now I will continue to try to solve my issue since I'm sure it is something fairly simple.  Simons monitoring program gives me all I need for recording multiple current readings and a single voltage reading.  My application is different than his but this works well for my needs.

Simon,
I will try what you suggested.  I am trying different things but it's interesting.  Seems that the current readings are working.  Only the voltage is not working.  I'm stumped as to why it gives a constant value without change but must be the driver.

Thanks,
George

harmongt01, 2018-01-12:
Simon,

Looking at these instructions:
sudo apt-get install git build-essential python-dev
cd ~
git clone https://github.com/adafruit/Adafruit_Python_ADS1x15.git
cd Adafruit_Python_ADS1x15
sudo python setup.py install

It says to go to the directory Adafruit_Python_ADS1x15.  Should there be a Python3 directory to use?  I can see the following directories in this folder:
Adafruit_Python_ADS1x15  
Adafruit_Python_PureIO  
Desktop  
pkg-mozilla-archive-keyring_1.1_all.deb  
Python-3.6.0  
Python-3.6.0.tgz  
tutorial_env

Anyway, I reinstalled but same thing with the voltage reading.  So I'll dig a little more.

simat, 2018-01-13:
Hi George,
So the Timeout problem is now fixed?

There is Only one ADS1x15 directory, the Adafruit code will work with versions 2 and 3 Python. If you are using python3 you have to install the software with Python3, thus the command sudo python3 setup.py install.

Typing just python will start whichever is the default version of python on your system which maybe python2.

Now you have installed  the ADS1x15 drivers go to the examples directory and run their test program simpletest.py to test the data coming from the voltage A to D.

Simon

simat, 2018-01-13:
Hi Bre55,

Thanks for your comments.

I haven't had any experience with MODBUS so thought it would be easier to decode the data packets being sent to and from the JBDTools software to the BMS boards and then simulate the JBDTools software to get the information I want from the BMS boards.

Do you know what the register map is for these boards and is there any sort of standard register map between different BMS manufactures?

Simon

bres55, 2018-01-15:
Hello Simon. Thanks for reply.
I am hoping not to distract you from the method you are using, and I may well have to explore that way also. 
I was provided this link from the supplier of my BMS
https://drive.google.com/file/d/0B3UXptx89r4NZ3VLTHlVS1ZGTTQ/view
It uses the method you use, but also refers to Modbus registers. But as I say, I have only had succes with holding register 3.  I am keen (maybe lazy) to stick with the Arduino and Modbus solution as I have a lot already connected and working. I shall continue to explore that for now and let you know if I have further success, And if you find anything in that line please let me know.
Best Regards
Michael.

simat, 2018-01-20:
Hi Bre55,

Thanks for the link. That fills in some holes in the information I have. I will update my documentation with the extra information when I get a chance.

If you are having trouble using Modbus I would just look at sending the correct data strings and reading the replies.

Simon

harmongt01, 2018-01-26:
HI Simon,

My timeout issue is still a problem.  I took some time from it but my solar charge controller is working well to discharge the batteries when needed.  

I may try to rebuild things from scratch again but I'm not sure that is the solution.  The OS is stable.  Really the only thing from the program on my end that does not work is the voltage reading.  So it's strange.  I still get the timeout error as before.  

I am sure it is something simple.  I would also like to start adding additional modules to measure and record my greenhouse, such as water temp, greenhouse temp, water pH, open vents and turn on fans, heat etc.  So I have a few things to figure out.

I may need to learn python better or use another language to build from.  What are your thoughts on that?

Thanks,
George

simat, 2018-01-30:
Hi George,

I assume you are not getting the timeouts with the A to Ds that are reading the current?

This is looking more and more like a problem with the voltage A to D chip. Might be worth changing the address of the Voltage A to D from x48 to x4B and testing it again.

Do you have a spare A to D you could try? Have you checked the wiring of the voltage A to D to see if it is OK?

I think Python is a good choice of language as there are a huge number of people using it. I think it is easier to use than say C and having the Python interpreter means you can easily test things. The major drawback is that it needs more computer resources than say C. This is not a problem with something like the Beaglebone but would be a problem with an Arduino.

Simon

There are lots of good interactive tutorials for Python, just google python tutorial.

harmongt01, 2018-02-01:
Hi Simon,

Right, I'm not getting the timeouts on the current AtoDs.  I did try swapping the the first AtoD which is the one reading the voltage but got the same result.  So I will try changing the address and see what that does.

The wiring hasn't changed and I did take readings at the voltage divider.  The voltage there does fluctuate as expected.

Thanks for the reply.  I will try changing the address to see what I get and I do intend to learn more with python.  There are a lot of things I'd like to automate and record in the next several years so it will be a fun learning curve.

-George

bres55, 2018-03-08:
Hello Simon.
Did/does your program write to the smartbms? 
I would really like to be able to switch the charge/discharge MOSFETs remotely.
I used the JBDTools pogram and used a 'serial port monitor' to watch the signal as I changed the mosfet using JBDT. 
it came out with
transmit                                                 reply
DD 5A 00 02 56 78 FF 30 77                DD 00 00 00 00 00 77
DD 5A E1 02 00 00 FF 1D 77                DD 00 00 00 00 00 77     
DD 5A 01 02 00 00 FF FD 77                DD 00 00 00 00 00 77
this was to turn the, charge mosfet off, discharge mosfet off, if allowed, other settings permiting.
the second line seems to be doing the business.

DD 5A E1 02 00 01 FF 1C 77           would turn charge mosfet on, discharge mosfet off, along with 1st and 3rd lines the same, same replies.
DD 5A E1 02 00 02 FF 1B 77          would turn charge mosfet off, discharge mosfet on, along with 1st and 3rd lines the same, same replies.
DD 5A E1 02 00 03 FF 1A 77          would turn charge mosfet on, discharge mosfet on, along with 1st and 3rd lines the same, same replies.
Notice its a 5A, to write, not the A5 to read.
I have written a crude arduino reader, that uses MODBUS protocol, that seems to work, but the write to a holding_register does not, write, as I would expect. So maybe not a full implementation of MODBUS.
Anyway, if you have had any succes in writing to the bms, I would be interested to hear.
Michael

simat, 2018-03-09:
Hi Michael,
The module that communicates with the BMS board is called `getbms.py` There is a function within the module `getbmsdat` that will send a command string and return the reply string in the variable 'data'

Can you make use of this?

It should be possible to use the alarms section of my code to trigger sending the mosfet on and off commands.

Simon

bres55, 2018-03-13:
Hello Simon.
I didn't get flagged you'd replied.
Thanks for pointer. I have been looking at the file you mention, and I am inching forward.
when you send the request code
('DD A5 03 00 FF FD 77')
are there any CR/LF, that you send? 0x77, is just a 'w', was wondering how the bms knows that's the end of data?
Michael

simat, 2018-03-13:
I wrote a better breakdown of the coms protocol on the endless spheres forum, here is the bulk of that post

This is the command sequence sent by the JBDTools software to obtain the battery information. It is a binary string which in HEX format is. I have added the EOLs for clarity.
DD A5 03 00 FF FD 77
DD A5 04 00 FF FC 77
DD A5 05 00 FF FB 77

First two bytes are the header, bytes 3 and 4 are the command, bytes 5 and 6 are the checksum, the last byte is the EOR.

The battery current along with other information that I haven't decoded is returned by the BMS board by the first command. The second command returns the individual cell voltages and the third returns the BMS name in ASCII format and maybe other information.

The reply packet is a binary string in HEX format is. I have added the EOLs for clarity
DD A5 00 1B 13 78 00 00 00 00 03 E8 00 00 22 C7 00 00 00 00 00 00 19 00 03 0C 02 0B 64 0B 5F FC 83 77
DD A5 00 18 10 39 10 3A 10 38 10 3A 10 3C 10 39 10 37 10 39 10 3B 10 3F 10 36 10 3A FC 74 77
DD A5 00 14 4C 48 2D 53 50 31 35 53 30 30 31 2D 50 31 33 53 2D 33 30 41 FB 39 77

The first two bytes are the header, bytes 3 and 4 are the length of the reply data in bytes, the data follows followed by the checksum, and finally the EOR.

The battery current is in bytes 7 and 8 of the first line

The second line are the individual cell voltages. Each voltage is a 16 bit number so 0x1039 gives a voltage of 4153mV
10 39 10 3A 10 38 10 3A 10 3C 10 39 10 37 10 39 10 3B 10 3F 10 36 10 3A
These converted to decimal in mV are 4153 4155 4152 4155 4157 4153 4151 4153 4156 4159 4150 4155

Third line is the BMS name.

bres55, 2018-03-13:
Thanks Simon for response. 
I am able to read, no probs. It's the writing to, using the 5A codes, that has me stumped, even then, I can write, but the changes are not actioned. I am off for a few days, maybe when I come back I'll see what I'm missing!
Michael

simat, 2018-03-14:
> I used the JBDTools pogram and used a 'serial port monitor' to watch the signal as I changed the mosfet using JBDT.
Hi Micheal
I can't see where you can turn the FETs off and on using JBDTools?

bres55, 2018-03-14:
Under 'OtherFunction' tab, there is an option for' MOS Control', where you can control the Charge and DisCharge MOSfets, as long as the batteries other conditions are met., for example,one would not want to turn on the Discharge Mosfet, if the batteries are already down to their limit.

bres55, 2018-03-14:
I know I am doing the right sequence, as I can use 'coolterm', connect to the BMS, open a 'send string' window, and send the three line of data to the BMS, and the MOSfet changes, but alas not with the arduino, and I am beginning to think there is some sort of hardware fault.

simat, 2018-03-15:
I did a quick test and tried sending your sequence

> DD 5A 00 02 56 78 FF 30 77......................DD 00 00 00 00 00 77
> DD 5A E1 02 00 00 FF 1D 77......................DD 00 00 00 00 00 77
> DD 5A 01 02 00 00 FF FD 77......................DD 00 00 00 00 00 77

via my software and confirm that  it doesn't work.
I will look further at this when I have some time.

bres55, 2018-03-15:
Thanks for trying that Simon.
I now have it working as expected. I am embarrassed to admit it was a typo all along, I put 255, should have been 225!!!!!!!!!! Argh!!!
I had an adage when I was working, blame the hardware, blame the software, blame yourself.. Well I had logic probes, puttting in delays, making sure signals were actually going through, I felt confident the software I was using was ok, so this morning I checked, using hard copy, my code, as sometimes that's the only way to see it differently, and found the offender. Sooo pleased to have cracked it before I go away. I'll update my github files in a week or so. Many thanks for helping to keep the ball rolling.

simat, 2018-03-16:
You have lost me. Where does the 255 (0xFF?)and 225(0xE1?) come in.
Is there a mistake with the three line command sequence?
I  have been asked to make up another off-grid system and wasn't going to use the onboard FETs but now think I might swap the FETs supplied with  some decent ones and use the FET control as another layer of protection.

bres55, 2018-03-21:
Hello Simon.
I have uploaded a new file to 
https://github.com/bres55/Smart-BMS-arduino-Reader
Find the Mega v4, and search through that for 'void call_control_on_on()' and other variants up to 'off_off'. I hope the code is logical from there and you will see the sequence required. I send the decimal values... hence the typo error...., but in most other ways I found that easier for me to write and follow. Yes I have enjoyed reading the 'Endless Sphere', thread on 'bluetooth BMS?'. I've learnt a lot from it and may consider up grading the mosfets and balance resistors at some stage, but will see how it goes for now. If you do, let us know which ones and how it goes. 
Michael

simat, 2018-05-17:
Hi Micheal,
How is progress with your project going?
I have reworked my BMS pcb to split the charge FETs from the discharge FETs using a Dremel to grind a slot in the copper between the two sets of FET drains.

The 30A BMS boards that I bought had 10 FETs (5 charge and 5 Discharge) on it. I moved the 5 charge FETs to the empty discharge locations on the other side of the PCB and installed two Vishay SUM80090E FETs which have a max Drain-Source voltage of 150V and an on resistance of ~10mOhms.

I have also run ~3.5mm squared cable from each of the FET sources and drains to take care of any problems running large currents through the inadequate PCB copper.

With this arrangement I can disconnect the solar panels from the SCC in the event of any cell voltage going too high and disconnect the inverter in the event of any cell going too low.

The original FETs configured as 10 discharge FETs run barely warm to touch with a discharge current of ~50A and the charge FETs the same with a panel current of ~10A. More tests to come during the next few days.

If you want any extra details let me know.

Simon

bres55, 2018-05-20:
Hello Simon. That all sounds good. You are way ahead of me on the hardware front. Since I installed a GTI there seems less urgency installing a bank of batteries, as I only have about 300, 18650s of any value. But I am hoping to install them soon. With the BMS, I eventually managed to find ways to turn on and off the charge and discharge fets with my arduino. That was my main aim, so I can turn the batteries, on and off remotely. And I can monitor the balancing, so if the 30ma is not enough, and I suspect it wont be, because of low quality cells, I may stick some mosfets and fans to help the balance discharge, using the BMS and arduino to monitor and control, sort of piggy back, on top of what the BMS is doing. It's interesting to hear how you are doing with the SUM80090E FETs, as there seem to be so many to chose from. But I am hoping I wont need to go down that road, unless I find a great suply of cells. Thanks for update.
All the best for now.

simat, 2018-06-13:
Hi George,
I think I might have worked out what your problem is.
Did you reflash the beagle with a new version of the kernel?
If the kernel version is newer than 3.8 I2C-1 is not enabled and initialised even though it shows up with the command `i2cdetect -l`. You can find out the kernel version with the command `uname -r`.

The fix for this is to edit the file `/boot/uEnv.txt`. In the entry in the section `###Custom Cape` change `<file8>` to  `BB-I2C1-00A0`. This will load and run the code that initialises I2C-1 at bootup.

Hope this is of use
Simon

harmongt01, 2018-06-13:
Hi Simon,

This sounds like a good idea.  I will check it.  

Thanks,
George

harmongt01, 2019-06-18:
Hi Simon,

Hope all is well for you.

I am trying to address the problem I had such a long time ago.  Needed to make some configuration change of my solar wiring and routing, etc.  Now I would like to get my monitoring back to running.

I did a git clone of your latest code.  I placed my battery.cfg file and trying to run I get:

root@beaglebone:/home/gharmon# python
Python 2.7.13 (default, Sep 26 2018, 18:42:22)
[GCC 6.3.0 20170516] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import batteries
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: No module named batteries
>>>

No module named batteries.  I must need to declare a value or something?

Thanks,
George

simat, 2019-06-19:
Hi George,
My later versions of the battery monitor need python3, or even better python3.7.
I don't think python 3.7 is available as a package for the Beaglebone, you would have to compile it yourself.
To install python3 use `apt-get install python3`
To start python3 type `python3`

I think you are in the wrong directory, git clone will create a new directory BatteryMonitor, you need to be in that directory to run my software.

Regards
Simon

harmongt01, 2019-06-19:
Hi,

I do remember you changed that and you’re right I was in the wrong directory, thank you.
New error:

root@beaglebone:/home/gharmon/BatteryMonitor# python3
Python 3.5.3 (default, Sep 27 2018, 17:25:39)
[GCC 6.3.0 20170516] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import batteries
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/gharmon/BatteryMonitor/batteries.py", line 26, in <module>
    import logger
  File "/home/gharmon/BatteryMonitor/logger.py", line 23, in <module>
    errfile=logging.FileHandler(config['files']['errfile'])
KeyError: 'errfile'
>>>

simat, 2019-06-20:
There have been several improvements and additions to the software since you were last using it which require new config information. I haven't been maintaining the various config files for the different configurations.

Probably the best thing would be for you to send your config file to me and I will modify it for you

harmongt01, 2019-06-20:
I’m not sure which config file I would need to update.  I attached my last battery.cfg but in the new clone I took I do not see a battery.cfg now so I guess it should be battery1cell.cfg?

simat, 2019-06-21:
There is only one config file used by my software, `battery.cfg` The other files are all data and log files.

All the .cfg files are just templates for different hardware configurations. If I am remembering your system correctly you are correct in thinking that battery1cell.cfg is the closest to what you want.

I think the easiest way to set things up for the new software would be to modify the  `battery,cfg` that you were  using when you had a working version of the software.

I can't see your battery config file attached to your last message?

harmongt01, 2019-06-22:
I tried attaching it to the email instead of through 

this site... my mistake.  Attached here and changed the extension from .cfg to .txt

[battery.txt](https://github.com/simat/BatteryMonitor/files/3316407/battery.txt)

closed 2020-04-11 by simat