simat/BatteryMonitor #13: Missing Modules

sandyeggoxj, 2018-02-26:
Here is the error I am receiving when trying to run this off the BMS data:
```
python3 batteries.py 3
Traceback (most recent call last):
  File "batteries.py", line 20, in <module>
    import Adafruit_GPIO as GPIO
ImportError: No module named 'Adafruit_GPIO'
```

sandyeggoxj, 2018-02-26:
If I do this:
```
#import Adafruit_GPIO as GPIO
import RPi.GPIO as GPIO
```

I then get this error:
```
~$sudo python3 batteries.py
Traceback (most recent call last):
  File "batteries.py", line 30, in <module>
    from getdata import Readings
  File "/home/pi/BatteryMonitor/getdata.py", line 27, in <module>
    class Readings:
  File "/home/pi/BatteryMonitor/getdata.py", line 37, in Readings
    ratio[i] = measured[i]/displayed[i]
IndexError: list index out of range
```

simat, 2018-02-26:
You will need to install the Adafruit GPIO modules.

If you are installing on the Beaglebone Black use the instructions [here](https://learn.adafruit.com/setting-up-io-python-library-on-beaglebone-black/installation-on-ubuntu)

sandyeggoxj, 2018-02-26:
Here is my battery.cfg file:

```
[files]
# Interface is the code to import the hardware interface driver
interface = 'from getbms import Raw'
# address of USB port for use with Bluetooth BMS board
usbport = '/dev/ttyUSB0'
# logfile is location and name of file used for logging all data
logfile = 'log'
# summaryfile is location and name of a summary of the data
summaryfile = '../summary'
# hoursummaryfile is the location of the hourly summary file
hoursummaryfile = '/dev/null'
# daysummaryfile is the location and name of daily summary file
daysummaryfile = '/dev/null'
# monthsummaryfile is the location and name of monthly summary file
monthsummaryfile = '/dev/null'
# yearsummaryfile is the location and name of yearly summary file
yearsummaryfile = '/dev/null'

[battery]
name = 'My Portable Battery'
numcells = 7
capacity = 20
ahloss = 0.1
vreset = 29.4  # voltage to reset DOD counter
ireset = 0.5  # current to reset DOD counter
ilowcurrent = 1.0 # C/20 lowcurrent voltage logging current

[VoltageInputs]
v1 = 'None'
v2 = 'None'
v3 = 'None'
v4 = 'None'
v5 = 'None'
v6 = 'None'
v7 = 'None'

[CurrentInputs]
iin1 = 'None'
iin2 = 'None'
iin3 = 'None'

[sampling]
# sampletime is time between samples in seconds
sampletime = 1.002
# samplesav is number of samples in running average
samplesav = 10

[calibrate]
currentgain = [-0.01, 1.0, 1.0]
currentoffset = [-0.0, 0.0, 0.0]
batvgain = 1
measured =  [0.0, 3.0, 6.0, 9.0, 12.0, 15.0, 18.0]
displayed = [0.0, 3.0, 6.0, 9.0, 12.0, 15.0, 18.0]
measureddelta =  [ 0.0, 3.000, 3.000, 3.000, 3.000, 3.000, 3.000]
displayeddelta = [ 0.0, 3.000, 3.000, 3.000, 3.000, 3.000, 3.000]

delta = [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
```

simat, 2018-02-27:
The problem is that you have numcell =7 and only 7 entries in the calibrate section. The first entry is for the zero volt readings.

Changing the [calibrate] entries to the following should fix the problem

```
measured =  [0.0, 3.0, 6.0, 9.0, 12.0, 15.0, 18.0, 21.0]
displayed = [0.0, 3.0, 6.0, 9.0, 12.0, 15.0, 18.0, 21.0]
measureddelta =  [ 0.0, 3.000, 3.000, 3.000, 3.000, 3.000, 3.000, 3.000]
displayeddelta = [ 0.0, 3.000, 3.000, 3.000, 3.000, 3.000, 3.000, 3.000]
```

sandyeggoxj, 2018-02-27:
Ok, I updated the config file to have the extra values like you noted above. Now I have this error:

```
pi@battery /var/www/html/BatteryMonitor $ python3 batteries.py
['batteries.py']
Traceback (most recent call last):
  File "batteries.py", line 173, in <module>
    deamon()
  File "batteries.py", line 47, in deamon
    while int(printtime) <= int(summary['current']['timestamp']):
KeyError: 'current'
```
Here is the section of the summary file it was referring to:
```
[current]
timestamp = '20140101000000 '
maxvoltages = [2.0, 2.0, 2.0, 2.0, 2.0, 2.0, 2.0, 20.0]
maxnocharge = [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
minnoload = [4.0, 4.0, 4.0, 4.0, 4.0, 4.0, 4.0, 30.0]
minvoltages = [4.0, 4.0, 4.0, 4.0, 4.0, 4.0, 4.0, 30.0]
deltav = [4.0, 0.0, 0.0]
amps = [500.0, -100.0, 0.0]
ah = [10000.0, 0.0, -1000.0, 0.0, 0.0, 0.0]
dod = [10000.0, 0.0, -1000.0, 0]
power = [0.0, 0.0, 0.0, 0.0]
```

simat, 2018-02-27:
That is odd? Looks like configparser is not parsing the summary file correctly.

I would compare your summary file with my template summary.bms.12cell.

The correct summary file is in the correct directory?

To debug you could put a debug statement just before line 47 of batteries.py
`print(summary,summary[current])`

sandyeggoxj, 2018-02-28:
The summary file should be in the main BatteryMonitor directory, correct? I have it called "summary" with no extension is that correct? 

Also, for the summary the only fields that should have more than 3-4 entries are: maxvoltages, maxnocharge, minnoload and minvoltages. Is that correct?

simat, 2018-03-01:
The only file that need to be in the same directory as where the code is being run from is the battery config file.

The location of the summary file is defined by the **summaryfile**  entry in the [files] section of the config file.

Looking at your config file
`summaryfile = '../summary'`
The ../ will mean that the program will look in the directory above the current directory. To look in the current directory you want `summaryfile = './summary'` or `summaryfile = 'summary'`

A key to the entries in the summary file can be found in [my Wiki here](https://github.com/simat/BatteryMonitor/wiki/Sample-Output)

Simon

sandyeggoxj, 2018-03-01:
Well I can tell you it definitely helps to reference the right summary file!!! Now batteries.py is actually running. It is still returning errors but we are getting closer. 

Here is my issue:
```
pi@battery /var/www/html/BatteryMonitor $ python3 batteries.py 2.5
['batteries.py', '2.5']
20180228200447
Traceback (most recent call last):
  File "batteries.py", line 171, in <module>
    deamon(float(sys.argv[1]))
  File "batteries.py", line 130, in deamon
    logsummary.update(summary, batdata)
  File "/var/www/html/BatteryMonitor/summary.py", line 77, in update
    summary['current']['ah'][6] = round(batdata.inahtot,2)  # current from solar etc
IndexError: list assignment index out of range
```

simat, 2018-03-02:
The ah variable in all the sections has seven elements not six, try this in your summary file

`ah = [10000.0, 0.0, -1000.0, 0.0, 0.0, 0.0, 0.0]`

sandyeggoxj, 2018-03-02:
Maybe I missed it somewhere in the documentation. How many items should each section have in the summary file? Some have more than others and I got confused.

simat, 2018-03-12:
Charged code to move Alarm section of the program to its own Class. The particular machine dependant version of the class can be specified in the config file.

closed 2018-03-12 by simat