simat/BatteryMonitor #16: Battery Monitor Open Source Project Inquiry

fordtheriver, 2018-10-30:
Hi Simon,

Greetings from Seattle, WA U.S.A! I'm a Mechanical Engineering student at the University of Washington interested in contributing to your open source project. 

I have custom built a Li-ion battery pack for my ebike and would like to gather more data about the battery's performance. My goal is to develop software for an Arduino to interface with a [generic Chinese BMS](https://www.aliexpress.com/item/15S-Li-ion-Battery-Intelligent-Smart-BMS-with-Bluetooth-function-and-PC-software-UART-communication-PCB/32876909159.html?spm=a2g0s.13010208.99999999.263.70483c00UKdgf7) (compatible with JBDTool software), a Hall-effect sensor (speedometer) and send power/speed/mileage data to a 20x4 LCD display via I2C. After extensive research into Smart BMS, it seems like there hasn't been much work done on reading their COM data with Arduino/Raspberry Pi. While a bluetooth app exists, I think it would be neat to have a LCD readout display on the bike.

From a glaze over your github, it looks like you've built software in Python for a BeagleBone to read data from a generic Chinese BMS (I believe the one I linked would work). It seems like our interests align somewhat so I'd be happy to collaborate my efforts here!

While I have some coding experience, I am very new to open source programming and Github and would be interested in hearing your thoughts on the matter.

Best regards,

Bryan Ford
BSME 2019
510 387 3498

simat, 2018-10-31:
Hi Bryan,
I can't find much info about the communication protocol either. This[ link](https://drive.google.com/file/d/0B3UXptx89r4NZ3VLTHlVS1ZGTTQ/view) gives some information on getting the basic data which is probably all you will need. I am in the process of reverse engineering the commands to change the setup parameters of the BMS. Should have this finished fairly soon, time permitting.

If power is not a problem I would stick with a Beaglebone or Raspberry Pi as it is far more flexible and easier to program that an Arduino. I can't see that power would be a problem if you are only using the device while the motor is being run or the battery is being charged.

GIT and github are a great versioning control system and way of sharing with others. I quite like this set of tutorials https://www.atlassian.com/git/tutorials but didn't bother with the interactive tutorials.

Regards
Simon Matthews

fordtheriver, 2018-11-01:
Hi Simon,

Thanks for your response. I'm looking at purchasing a Raspberry Pi Zero for this project (It'll be on an ebike so size is a big constraint). I'll take a look at the Git tutorials as well. Would you recommend any software for a Raspberry Pi  that you've found helpful for this project? Also, would you recommend any quality Raspberry Pi tutorials?

Again, thank you for the support and I'm excited to contribute to this project!

Regards,
Bryan

simat, 2018-11-04:
Hi Bryan,
I think the Raspberry Pi Zero will have enough horse power to run python and my software. Will be interesting to see if this is the case.

The Pi Zero runs a variant of the Debian distribution of Linux. Have you had any experience with using Linux?

The instructions on the Raspberry Pi website https://www.raspberrypi.org/products/raspberry-pi-zero/ seem pretty good and are backed up with the Raspberry Pi user forum. If you get stuck I am happy to help out.

It would be great if you could document what you are doing on my WIKI or maybe in some other form.

Simon

fordtheriver, 2018-11-07:
Hi Simon,

Thanks for sending additional material, I do not have any experience with Linux. 

After some initial research, I'm have some concerns about using the Raspberry Pi over an Arduino. It looks I can't immediately shut off power to the Pi and need to go through a boot/reboot process every time I want to use the bike (assuming that I would be able to tell the Pi to boot/reboot with a switch, because I won't have a keyboard/mouse on the bike).

It looks like you can put the Pi into "sleep" mode with an external switch and draw at least 1W of power as it sleeps. While this solution seems to mitigate long boot/reboot times, 1W is a pretty significant chunk of my ~600 W-hr battery pack and this ignores any loss of a buck converter to go from ~55V to 5V. 

Are you aware of any ways to quickly turn on/off the raspberry pi with an external switch? Or significantly reduce the amount of power it consumes in "sleep" mode?

Bryan

simat, 2018-11-09:
Hi Bryan, a quick google search brings up that  you can get the Pi zero to boot in around four seconds, I assume you have to strip out all the network setup and other things or at least put this after starting your application. Not ideal but doable. I also saw a post that says the Pi zero uses about 100mA (0.5W) when idle.

I don't think there would be a problem setting up a switch to start and stop the Pi.

If all you wanted to do is display the speed and battery state info I would have thought an Arduino would do the job although I have very little experience with the Arduino. It might be a steep learning curve for you if you haven't done any programming in C. hopefully you would be able to find libraries that would do most of the hard work like reading from the BMS unit, decoding the data and writing to a display of some sort. Have you looked on the [Endless Spheres Forum](https://endless-sphere.com/forums/viewforum.php?f=14&sid=3f8d1778ed35a373018b69ad8b45326e)
Simon

closed 2020-04-11 by simat