simat/BatteryMonitor #17: CRC Error

zaziki23, 2019-10-11:
I tried your bmstest.py script, but i'm not able to read data from my bms. It always sys CRC Error
The connection is open on /dev/rfcomm1 while this test.
Can you help me with that? 

`root@raspberrypi:/opt/anotherBMS# python3 bmstest.py
['bmstest.py']
Enter BMS port address option [3]
(1) /dev/ttyUSB0
(2) /dev/ttyUSB1
(3) other
Enter Command>/dev/rfcomm1
Traceback (most recent call last):
  File "bmstest.py", line 244, in <module>
    main()
  File "bmstest.py", line 149, in main
    port=int(getcmd())
ValueError: invalid literal for int() with base 10: '/dev/rfcomm1'
root@raspberrypi:/opt/anotherBMS# ^C
root@raspberrypi:/opt/anotherBMS# python3 bmstest.py
['bmstest.py']
Enter BMS port address option [3]
(1) /dev/ttyUSB0
(2) /dev/ttyUSB1
(3) other
Enter Command>3
Enter port name>/dev/rfcomm1
Enter option
(1) Load all config data from BMS to memory
(2) Read all config data from disk to memory
(3) Write all config data from memory to BMS
(4) Write all config data from memory to disk
(5) Dump all config data in memory
(6) Dump raw config data in memory
(7) Read/Write config item in memory
(8) Read/Write single register in memory and on BMS PCB
(9) Read BMS data
(10) Switch charge/discharge FETs
(11) Calibrate BMS
Enter Command>9
port is:  /dev/rfcomm1
command= b'dda50300fffd77'
CRC Error
reply= b''
command= b'dda50300fffd77'
CRC Error
reply= b''
V=0 I=0 bal=0 state=0 fets=0
b''
command= b'dda50400fffc77'
CRC Error
reply= b''
b''
[0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
command= b'dda50500fffb77'
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 501, in read
    'device reports readiness to read but returned no data '
serial.serialutil.SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "bmstest.py", line 244, in <module>
    main()
  File "bmstest.py", line 210, in main
    getdat(port)
  File "bmstest.py", line 86, in getdat
    dat = bmscore.getbmsdat(ser,command)
  File "/opt/anotherBMS/bmscore.py", line 58, in getbmsdat
    reply = port.read(4)
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 509, in read
    raise SerialException('read failed: {}'.format(e))
serial.serialutil.SerialException: read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
root@raspberrypi:/opt/anotherBMS#
`

simat, 2019-10-11:
It looks to me like you are getting no reply from /dev/rfcomm1
Are you sure this port is setup correctly?
I just had a look at the documentation for the Raspberry Pi. I didn't know you could get to the bluetooth as a serial port. I have never tried doing this.

Two options I have tried are connecting the BMS board via a USB to serial dongle and connecting the serial port directly from the BMS board to one of the Beaglebone Black serial ports. Can't see why this wouldn't work on the Pi as well

I have been doing quite a bit of work on bmstest recently, have you done a git pull to update to the latest?

zaziki23, 2019-10-11:
well, you can create a serial emulation via bluetooth with the command rfcomm.

> root@raspberrypi:/opt/anotherBMS# rfcomm connect /dev/rfcomm1 AA:BB:CC:A1:23:45 1
Connected /dev/rfcomm1 to AA:BB:CC:A1:23:45 on channel 1
Press CTRL-C for hangup

i think that should work.

> I have been doing quite a bit of work on bmstest recently, have you done a git pull to update to the latest?
i pulled this morning.

So you never talked to the bms via bluetooth ? hm. OK, i thought that was the case.

So you mean via a cheap rs232 usb converter ? hm, ok i have to look where i can find the pinout for that.

I read from some placed that it is possible to read them via bluetooth - which i liked.

e.g.
[here](https://mono.software/2018/11/15/multiple-bms-monitor/)
[and here](https://github.com/sw-home/FHEM-BluetoothSmartBMS)

simat, 2019-10-12:
I will have a play with the bluetooth when I get home, very interesting.

Yes a cheap 3.3v serial to USB converter is the way I usually do it. 

You could use the [Pi's onboard uarts](https://www.raspberrypi.org/documentation/configuration/uart.md) directly.

zaziki23, 2019-10-13:
i opened the bms this morning but i can not figure out where a uart pinout whout be.
i get the feeling that i have a complete different bms.

can you please take a look at this photos.
[image]

[image]

simat, 2019-10-14:
Yes, that is the wrong board

I had a play with using rfcomm with my BMS board. 

rfcomm bind seems to work OK, pyserial opening port doesn't give an error but when you try to write to the port you get 

> Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.7/site-packages/serial/serialposix.py", line 537, in write
    n = os.write(self.fd, d)
OSError: [Errno 5] Input/output error

Has anyone else had any luck using rfcomm and pyserial?

zaziki23, 2019-10-15:
can you provide me a link to a suitable product?

simat, 2019-10-16:
https://www.aliexpress.com/item/32733414980.html
I have checked the permissions, should be OK. I think it is something more fundamental

closed 2020-04-11 by simat