simat/BatteryMonitor #18: Error with logger.py

robermf, 2019-12-01:
Hello. 
First of all say thank you for the software. I was really looking for something like this. 

I have the following problem when I try to run python3 batteries.py:

--------
pi@raspberrypi:~/BatteryMonitor $ python3 batteries.py
Traceback (most recent call last):
  File "batteries.py", line 26, in <module>
    import logger
  File "/home/pi/BatteryMonitor/logger.py", line 23, in <module>
    errfile=logging.FileHandler(config['files']['errfile'])
KeyError: 'errfile'
---------

When I run bmstest.py I can connect to the BMS and get the data (USB attached). The summary file is store in right folder according with battery.cfg
-------------
[files]
interface = 'from getbms import Raw'
# logfile is location and name of file used for logging all data
logfile = 'data/log'
# summaryfile is location and name of a summary of the data
summaryfile = 'data/summary'
# hoursummaryfile is the location of the hourly summary file
hoursummaryfile = '/dev/null'
# daysummaryfile is the location and name of daily summary file
daysummaryfile = '/dev/null'
# monthsummaryfile is the location and name of monthly summary file
monthsummaryfile = '/dev/null'
# yearsummaryfile is the location and name of yearly summary file
yearsummaryfile = '/dev/null'
---------------
(also tried writing the full path)

Any idea about what am I doing wrong?

Thank you again. 

Best regards. 

PS: I have checked in Issues section but I didn't see something like this.

simat, 2019-12-02:
Some of the config files I provided had not been updated to include the location of the a new error log file.
To fix just add the error file location in the [files] section of the config

# errfile is location and name of error log
errfile = 'data/errorlog'

I have updated all the config files

robermf, 2019-12-02:
Thank you for your help simat. 

I make the changes (also tried using git clone again) and I got the next error: 

pi@raspberrypi:~/BatteryMonitor $ python3 batteries.py [0]
Traceback (most recent call last):
  File "batteries.py", line 26, in <module>
    import logger
  File "/home/pi/BatteryMonitor/logger.py", line 25, in <module>
    alarmfile=logging.FileHandler(config['files']['alarmfile'])
KeyError: 'alarmfile'

So I add "alarmfile = 'data/alarmlog'" to battery.cfg. Then, the next I got is: 

pi@raspberrypi:~/BatteryMonitor $ python3 batteries.py [0]
['batteries.py', '[0]']
Traceback (most recent call last):
  File "batteries.py", line 161, in <module>
    deamon(float(sys.argv[1]))
ValueError: could not convert string to float: '[0]'
pi@raspberrypi:~/BatteryMonitor $ python3 batteries.py    
['batteries.py']
Traceback (most recent call last):
  File "batteries.py", line 163, in <module>
    deamon()
  File "batteries.py", line 49, in deamon
    batdata = Readings()  # initialise batdata after we have valid sys time
  File "/home/pi/BatteryMonitor/getdata.py", line 84, in __init__
    for i in config['Interfaces']:
KeyError: 'Interfaces'

I have been trying different things without success. Any suggestion?

Best regards.

simat, 2019-12-02:
try python3 batteries.py 0
You need an interfaces section in the config file something like

```
[Interfaces]
#bms = 'bms(LH-SP15S001-P15S-30A)' # string in brackets is harware name, from read command 05
```
What equipment are you using the software with?

robermf, 2019-12-02:
One step closer :)
I was using as example battery12cellBMS.cfg wich has a few different parametres from battery16cellBMS.cfg.
Now I am using 16cell, but with the 12cell structure and have some progress.

This is my battery.cfg:

```
[files]
logfile = '/home/pi/BatteryMonitor/data/log'
summaryfile = '/home/pi/BatteryMonitor/data/summary'
errfile = '/home/pi/BatteryMonitor/data/baterrlog'
alarmfile = '/home/pi/BatteryMonitor/data/alarmlog'
hoursummaryfile = '/home/pi/BatteryMonitor/data/hoursummary'
daysummaryfile = '/home/pi/BatteryMonitor/data/daysummary'
monthsummaryfile = '/home/pi/BatteryMonitor/data/monthsummary'
yearsummaryfile = '/home/pi/BatteryMonitor/data/yearsummary'

[logging]
data = "self.printtime+batdata.vbat+batdata.vdelta \
	+batdata.iall+batdata.soctxt+batdata.socadjtxt+str(batdata.pip.rawdat)"

[battery]
name = 'BMS01'
numcells = 12
capacity = 180
overvoltage = 4.200
undervoltage = 3.500
ahloss = -0.005
vreset = 51.5  # voltage to reset DOD counter
ireset = 5.0  # current to reset DOD counter
ilowcurrent = 9.0 # C/50 lowcurrent voltage logging current

[Interfaces]
pip = 'pip(92931710100542)' # number in brackets is SN of PIP)
bms = 'bms(JBD-SP15S008C-P13S-100A-B)' # string in brackets is harware name, from read command 05

[Ports]
pipport = '/dev/ttyU*' # where to look for pipport
bmsport = '/dev/ttyUSB0' # where to look for bmsport

[VoltageInputs]
v01 = 'self.bms.rawdat["V01"]'
v02 = 'self.bms.rawdat["V02"]'
v03 = 'self.bms.rawdat["V03"]'
v04 = 'self.bms.rawdat["V04"]'
v05 = 'self.bms.rawdat["V05"]'
v06 = 'self.bms.rawdat["V06"]'
v07 = 'self.bms.rawdat["V07"]'
v08 = 'self.bms.rawdat["V08"]'
v09 = 'self.bms.rawdat["V09"]'
v10 = 'self.bms.rawdat["V10"]'
v11 = 'self.bms.rawdat["V11"]'
v12 = 'self.bms.rawdat["V12"]'
vbat = 'self.bms.rawdat["V12"]'

[CurrentInputs]
iin1 = 'self.bms.rawdat["Ibat"]'
iin2 = 'self.pip.rawdat["PVI"]'
iin3 = 'self.current[0]-self.current[1]'

[TemperatureInputs]
tin1='self.bms.rawdat["T1"]'
tin2='self.bms.rawdat["T2"]'

[BalanceFlags]
balf01 = 'self.bms.rawdat["Bal"] & 1'
balf02 = 'self.bms.rawdat["Bal"] >> 1 & 1'
balf03 = 'self.bms.rawdat["Bal"] >> 2 & 1'
balf04 = 'self.bms.rawdat["Bal"] >> 3 & 1'
balf05 = 'self.bms.rawdat["Bal"] >> 4 & 1'
[... 92 lines cut ...]
(8) Read/Write single register in memory and on BMS PCB
(9) Read BMS data
(10) Switch charge/discharge FETs
(11) Calibrate BMS
Enter Command>9
command= b'dda50300fffd77'
CRC Error
reply= b''
command= b'dda50300fffd77'
reply= b'114d000000001770000027250000000000021000010c020b660b68'
V=4429 I=0 bal=0b0 state=0b10 fets=0b1
Cell Undervoltage
b'114d000000001770000027250000000000021000010c020b660b68'
command= b'dda50400fffc77'
reply= b'0fa40fa50fa50fa50fa40fa90f760fa40fab0faa0fa10119'
b'0fa40fa50fa50fa50fa40fa90f760fa40fab0faa0fa10119'
[4.004, 4.005, 4.005, 4.005, 4.004, 4.009, 3.958, 4.004, 4.011, 4.01, 4.001, 0.281, 0.0, 0.0, 0.0]
command= b'dda50500fffb77'
reply= b'4a42442d5350313553303038432d503133532d313030412d42'
b'4a42442d5350313553303038432d503133532d313030412d42'
```

Thank you for the help and sorry for bothering. 

Regards.

simat, 2019-12-16:
Sorry, I missed your last post.

My fault again. Added code that made the overall battery voltage separate from the last cell voltage but didn't update the config files

To hopefully fix, in the [calibrate] section of the config file add 36.0 to the end of 'measured' and 'displayed' and add 3.0 to 'measureddelta' and 'displayeddelta' and 0.0 at the end of 'delta'

robermf, 2019-12-17:
THANK YOU VERY MUCH!

Now it is working! :))) 

I had to modify the summary file (I used 12 cell config) to add in current section:

`tmax = [0.0, 0.0, 0.0, 0.0]
tmin = [100.0, 100.0, 100.0, 100.0]
baltime = [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
state = ['No Sun']
batpwr1hrav = [0.0]`

and to the rest sections: 
`tmax = [0.0, 0.0, 0.0, 0.0]
tmin = [100.0, 100.0, 100.0, 100.0]
baltime = [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]`

And all is working perfect! It will really helpful for me. 
Now I am going to get de data from battery.php to Home Assistant to get alerts and another automations. Also I have to try to run another instance, or try Raspberry Pi Zero W, becasue I would like to manage 8 BMS in total.

Simat, thank you again. 

Best regards!

closed 2020-04-11 by simat