simat/BatteryMonitor #19: Issue with BMS Hardware Name

lekr74, 2020-06-19:
Hello,

I'm trying to use your code with a chinese 8S bluetooth BMS
I can perfectly read data with the bmstest.py using /dev/ttyUSB0

The thing is that, when I try to use batteries.py, it seems to need a hardware name for the BMS to connect to.

I saw in [this comment](https://github.com/simat/BatteryMonitor/issues/18#issuecomment-560412766) that I need to put this in battery.cfg : 
```
[Interfaces]
bms = 'bms(LH-SP15S001-P15S-30A)' # string in brackets is harware name, from read command 05
```

but i can't figure out how to find the hardware name.
I tried with SN or Model, but the BMS is answering ABCD for both SN and Model.

Could you explain me how to find the hardware name ?

Thanks in advance !

simat, 2020-06-20:
I think the problem here is when using bmstest you are not loading the config data from the BMS to memory before trying to look at the data.

Issue a 1 command to load the data from the BMS before either a 5 or 8 command to then show the data loaded from the BMS. You want the register called 'Model'.

Alternatively you could use the bluetooth phone app to get the Model number.

Regards
Simon

lekr74, 2020-06-20:
Hello
Thanks a lot for your fast answer !

So indeed it was the problem, I did it and got the model.
I then did disable the line 162 & 163 of getdata.py as it was trying to read data from a non existent pip.
Now I get this : 

```
pi@raspberrypi:~/BatteryMonitor $ python3 batteries.py 0
['batteries.py', '0']
Traceback (most recent call last):
  File "batteries.py", line 162, in <module>
    daemon(float(sys.argv[1]))
  File "batteries.py", line 49, in daemon
    batdata = Readings()  # initialise batdata after we have valid sys time
  File "/home/pi/BatteryMonitor/getdata.py", line 124, in __init__
    self.batcurrentav = self.current[-3]
IndexError: list index out of range
```

Maybe there is another problem with my battery.cfg ?
I attach it in case it's that

thanks again

regards,

Alexandre
[battery.txt](https://github.com/simat/BatteryMonitor/files/4807563/battery.txt)

simat, 2020-06-20:
The program needs at least three current inputs, 
[-3] is the battery current
[-2] is the charge current
[-1] is the load current
You can make these equal to zero, or calculate them somehow, some examples
iin2=0
iin3=self.current[0]-self.current1 # self.current[0]=iin1, self.current[1]=iin2 ...

Rather that changing Getdata just leave the [Status] section of the config file blank i.e.
[Status]

[sampling] ...

Simon

lekr74, 2020-06-20:
ok
I changed the [Status section] and changed back Getdata

I tried 
`iin3=self.current[0]-self.current1 # self.current[0]=iin1, self.current[1]=iin2 ...`
but it didn't work, and same thing using 0 for both
So I did use this : 
```
[CurrentInputs]
iin1 = 'self.bms.rawdat["Ibat"]'
iin2 = 'self.bms.rawdat["Ibat"]'iin3 = 'self.bms.rawdat["Ibat"]'
```
and it seems to be better, as I'm getting a date when running the script … but another error too :

``pi@raspberrypi:~/BatteryMonitor $ python3 batteries.py 0
['batteries.py', '0']
20200620142416 
Traceback (most recent call last):
  File "batteries.py", line 162, in <module>
    daemon(float(sys.argv[1]))
  File "batteries.py", line 116, in daemon
    logsummary.update(summary, batdata)
  File "/home/pi/BatteryMonitor/summary.py", line 87, in update
    summary['current']['ah'][6] = round(batdata.inahtot,2)  # current from solar etc
IndexError: list assignment index out of range``

By the way, my goal is rather simple, I just want to get the data from the BMS to Influxdb in order to graph them with grafana …

thanks

Alexandre

simat, 2020-06-21:
Not sure which version of the summary file you are using. 
Some of the template summary files are out of date and need to be modified. summary.bms.2inverters.15cells is a good guide for any new summary file.
If you get stuck let me know which summary template you used and I will modify it for you.

A better approach might be to write a python program and use the getdat()) function from bmstest.py to get the data from the BMS boards and pass it on to influxdb using one of the python influxdb libraries.

Simon

closed 2020-06-28 by simat